qmk/qmk_firmware #1000: tap dance for MO() and TG() based on count? Or custom oneshot?

rclasen, 2017-01-10:
Hi,

I do have tap dance working for simple keys... but fail with something like this:

ACTION_TAP_DANCE_DOUBLE( MO(1), TG(1) )

Is this supposed to work?

Instead I'd be happy with OSL() ... but due to the timeout bug and lack of possibility to easily deactivate the oneshot state, I don't get along with it... no matter if I use oneshot toggle or not.

thanks

pvinis, 2017-01-11:
its possible. you have to make a custom key, then assign that somewhere and have it work as a tapdance key. then you make the action of the tapdance as you want. i have it so if i tap once, i go to one layer. if i tap twice, i go to another layer. then while in any of the two layers, if i tap once i go back to the initial layer.
you can check that here https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/ergodox/keymaps/pvinis/keymap.c#L474 and here https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/ergodox/keymaps/pvinis/keymap.c#L448-L470 and here https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/ergodox/keymaps/pvinis/keymap.c#L25.
(i think i have a bug where if i tap once, then tap twice, im stuck in one of the layers and cant go back, but eh, i dont do that. i will fix it later.)

rclasen, 2017-01-11:
Thanks for sharing. I see... you coded up the toggle yourself. So in fact using the "easy" TG() by itself might be incompatible with tap dance.

I glanced at the tapdance code in process_tap_dance.c... and I got the impression, it's just handling key presses... the release appears to be ignored (or I missed it). That'd be a show stopper for MO() like layer switching while holding down a key.... hmm...

blakedietz, 2017-01-15:
@pvinis: The reason why you have a bug when you tap three times is because you don't have a default case statement for your switch block.

See this for an example: https://github.com/blakedietz/qmk_firmware/blob/hhkb/keyboards/hhkb/keymaps/blakedietz/keymap.c#L167-L169.

Thanks for that snippet. It helped me to implement my own layer switching for my hhkb.

pvinis, 2017-01-15:
@blakedietz hm, i thought about it, but its not that actually. when i tap once, i go to layer `sysctl`. then i tap twice, and i go to layer `mouse` without checking. so i think i would either put a switch in the `case 2:`, or better, maybe instead of `trns` in the two non-`base` layers, i should have an other key thats justs goes turns the current layer off.

skullydazed, 2017-06-27:
I'm closing this to clean up our open issues, but if you still have questions @rclasen please reopen this or open a new issue.

closed 2017-06-27 by skullydazed

xcode-manolo, 2018-08-03:
ALL THE LINKS ARE 404_ERRORS NOW !!!

Where can we get those sources ?
I need to implement this functionality !!!

Thanks anyone can help !!!

noroadsleft, 2018-08-03:
@xcode-manolo 

Don't know how to find the link from @blakedietz's Jan 14, 2017 post, but...

Links from @pvinis's post, Jan. 11, 2017:

- https://github.com/qmk/qmk_firmware/blob/3a453a03ed5f029bb9cf23955e87789245d39a22/keyboards/ergodox/keymaps/pvinis/keymap.c#L474
- https://github.com/qmk/qmk_firmware/blob/3a453a03ed5f029bb9cf23955e87789245d39a22/keyboards/ergodox/keymaps/pvinis/keymap.c#L448-L470
- https://github.com/qmk/qmk_firmware/blob/3a453a03ed5f029bb9cf23955e87789245d39a22/keyboards/ergodox/keymaps/pvinis/keymap.c#L25

Hope it helps.

pvinis, 2018-08-03:
sorry for the delay. thanks @noroadsleft. enjoy @xcode-manolo. I'm updating my keymap this week too :)