qmk/qmk_firmware #998: planck-4.1-398-g1585fc-dirty

arianvp, 2017-01-10:
My keyboard was stuck and then it suddenly typed out `planck-4.1-398-g1585fc-dirty`

I cannot find this magic string back in the source code anywhere. Where is it defined and what does it mean?

fredizzimo, 2017-01-10:
It looks like you manager to activate a layer containing the "version" key. You didn't specify which keymap you are using, so I can't tell which layer you activated.

It's supposed to print `keyboard`-`keymap`-`git version`, but it seems like there's a bug so the `keyboard` and `keymap` are not passed correctly. It's most likely due to the makefile refactoring I did last summer. So I can take a look at it at some point.

jackhumbert, 2017-02-07:
With the new versioning we've done, this should print something a little more meaningful :) but yeah, this is a feature in the Ergodox keymaps.

closed 2017-02-07 by jackhumbert

fredizzimo, 2017-03-26:
@jackhumbert, has the actual bug been fixed? Notice how it prints `planck`  instead of `ergodox`

jackhumbert, 2017-03-27:
Yeah, it just prints the most recent tag (which was planck-4.1). We've since introduced actual versioning with tags, so this number is a little more useful :)