qmk/qmk_firmware #1001: Satan: Last column (backspace/enter/shift/etc.) doesn't work

alexyoung, 2017-01-11:
I know there was issue #499 where the fourth column didn't work, and it was found to be a soldering issue. I have a Satan board here that works fine with EasyAVR, but the last column doesn't work when I use qmk. I've tried changing the last column's pin, [it's B3 in EasyAVR](https://github.com/dhowland/EasyAVR/blob/60f26ea9eccc40d51cc900780d256621165663d7/keymapper/easykeymap/boards/gh60_satan.py#L43) and that works for me.

I've tried building with LEDs disabled (I don't have any on this board anyway), and I've toggled some of the options in the Makefile, like NKRO.

Is anyone using qmk successfully with their Satan board? If so close the issue I guess, it's probably something obscure to do with my build.

Dbroqua, 2017-01-11:
Hello,

Have you a keyboard-layout-editor of the layout you want ?

I have a GH60 Satan with MX HHKB layout and all works fines.

alexyoung, 2017-01-12:
It's only ANSI 60% with ctrl instead of caps, Fn instead of left ctrl, where Fn switches to a second layout that does media control, F1-F12, Delete, ~/`, etc.

http://bit.ly/2ja6YE0

Dbroqua, 2017-01-13:
Have you try to flash it with the default keymap ?

fredizzimo, 2017-01-15:
@alexyoung As far as I can see it's `B3` here too https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/satan/config.h. So what exactly did you change?

alexyoung, 2017-01-16:
I kept the defaults and it doesn't work with that last row, I changed a few things just to experiment.

These are the keymaps I've tried:

* default
* poker
* dbroqua

My last config.h changes were:

* `#define BACKLIGHT_LEVELS 0`
* `// #define NO_DEBUG`
* `// #define NO_PRINT`

This is my preferred keymap that I've been trying to get to work: https://gist.github.com/alexyoung/7b32853f743b0c05edf59d38818b237b

fredizzimo, 2017-01-17:
OK, then I don't think it's a software problem.

Have you tried reflashing EasyAVR again after you encountered the issue? There might be some loose contact somewhere, which have caused it fail, unrelated to QMK. If that has happened then EasyAvr should probably fail too. 

Otherwise it's probably a similar issue as in #499, a short-circuit somewhere.

So with this in mind, I would disconnect the keyboard and check that all connections work properly with a multimeter. You can start by testing the continuity between the pin that is connected to the last column and all other pins on the controller. None of them should be connected, otherwise you have a short-circuit.

You can also test that the pin really is connected to all the switches on the last column. But if that's not the case, then I don't see how EasyAVR would work either. And finally that if you press a switch, that it is connected to the correct row pin, but for that you need to do a [diode test](http://en-us.fluke.com/training/training-library/test-tools/digital-multimeters/how-to-test-diodes-using-a-digital-multimeter.html). It's very improbable that this test would fail though, since the whole column is dead for you.

alexyoung, 2017-01-26:
Thanks for the suggestions, I'll get the multimeter out and keep testing!

closed 2017-06-27 by skullydazed