qmk/qmk_firmware #1008: Let's Split: pin settings in keymaps/serial/config.h override rev2/config.h

TerryMathews, 2017-01-15:
I've got a rev2 Let's Split using serial instead of i2c. The subproject logic that QMK is using is having it include keymaps/serial/config.h over rev2/config.h, and the pin mappings in keymaps/serial/config.h are for rev1.

I fixed this in my local copy by overriding the pin mappings in keymaps/serial/config.h to rev2, but that's obviously not a good long-term solution. I don't have a suggestion on how to fix this at this time, I haven't played around with the new subproject logic enough.

jackhumbert, 2017-01-15:
You should be able to override those settings in your keymap's Makefile and config.h - you can use `#undef` in the config.h. 

It should be including both the project and subproject config.h, though (through includes). The pin mappings shouldn't be in the keymap's config.h - just where they're needed in the (sub)project's.

TerryMathews, 2017-01-15:
This isn't my code - if it were my project I'd probably switch from rev1/rev2 subprojects and serial/i2c keymaps and go to rev2-serial, rev2-i2c etc.

I tried adjusting the source in the manner you explained, but compiling fails at matrix.c.

I don't know if you want to tag in the maintainer or not, I can tell you the code that is spit out does not match the file name. lets_split_rev2_serial.hex uses the matrix pinouts of rev1. Since there is a matrix difference between 1 and 2, you end up with code that doesn't work.

TerryMathews, 2017-01-15:
I played around with it some more:

To me, it appears that the crux of the issue comes from compiling several of the support functions that are in the top level directory:
* matrix.c
* split_util.h and .c

Both invoke MATRIX_ROWS/COLUMNS and MATRIX_ROW_PINS/COLUMN_PINS and the compile is breaking when I move those defines into their respective rev1/rev2 directories because the make function hasn't traversed into the subproject at that point.

It looks to me with the way the build project traverses the source tree, that the support libraries that depend on the pins assignments need to be duplicated and moved within the subproject folders.

fredizzimo, 2017-01-15:
The makefile logic is working as it should, but the problem is that the keymap specific config file is defining too much, and essentially redefining what should already be defined by the keyboard itself.

The fix should be to remove all extra stuff and include `../../config.h` instead of `config_common.h`. The `../../config.h` file includes the correct revision specific config file which defines the pins and other stuff correctly.

TerryMathews, 2017-01-15:
Sorry, I never meant to imply that the makefile logic was broken. I know
it's an issue with the source files.

fredizzimo, 2017-01-16:
@TerryMathews, don't worry, I was probably reading too much between the lines. Now when I read your comments again, there's really nothing that implies that.

I still think that the changes I suggested should work if done correctly, but without trying it myself there's probably something I'm missing. Could you please share the changes that you made, and what compilation error you get? That would make the problem easier to see and discuss.

closed 2017-01-23 by TerryMathews