qmk/qmk_firmware #1011: Problem with Ableton and Max/MSP on Sierra

szymonkaliski, 2017-01-16:
I know this looks strange, and I have problems pinpointing the issue, so here it goes: I have Ergodox from Falbatech, with QMK firmware working nicely (for almost a year now), recently I upgraded to Sierra, and now every time I run Max/MSP or Ableton the keyboard stops working, red LED is turned on, and I can only get it back to work if I quit Max and Ableton, unplug, and plug the keyboard again.

I tried looking in the Console when I start running one of those apps, and I'm getting little bit of info:

```
default	11:18:34.714281 +0100	icdd	#ICDebug - 51:{ICWiredBrowser.m} (USB terminate)
default	11:18:34.714350 +0100	icdd	#ICDebug - 344:{ICWiredBrowser.m} (--> USB Terminate)
default	11:18:34.714633 +0100	icdd	#ICDebug - 363:{ICWiredBrowser.m} (Reclaiming Resource: ErgoDox EZ)
default	11:18:34.714692 +0100	icdd	#ICDebug - 363:{ICWiredBrowser.m} (Reclaiming Resource: ErgoDox EZ)
default	11:18:34.714976 +0100	icdd	#ICDebug - 363:{ICWiredBrowser.m} (Reclaiming Resource: ErgoDox EZ)
default	11:18:34.715050 +0100	icdd	#ICDebug - 363:{ICWiredBrowser.m} (Reclaiming Resource: ErgoDox EZ)
default	11:18:34.715138 +0100	icdd	#ICDebug - 363:{ICWiredBrowser.m} (Reclaiming Resource: ErgoDox EZ)
default	11:18:34.715275 +0100	icdd	#ICDebug - 363:{ICWiredBrowser.m} (Reclaiming Resource: ErgoDox EZ)
default	11:18:34.715315 +0100	icdd	#ICDebug - 509:{ICDDMessageCenter.m} (-Rem ErgoDox EZ - 0x0/0x0/0x0 - 0x41130000 - ICDeviceDescriptionUpdated)
default	11:18:34.715329 +0100	icdd	#ICDebug - 363:{ICWiredBrowser.m} (Reclaiming Resource: ErgoDox EZ)
default	11:18:34.715392 +0100	icdd	#ICDebug - 371:{ICWiredBrowser.m} (13 USB Descriptions Managed)
default	11:18:34.715438 +0100	icdd	#ICDebug - 373:{ICWiredBrowser.m} (<-- USB Terminate)
default	11:18:34.715532 +0100	icdd	#ICDebug - 509:{ICDDMessageCenter.m} (-Rem ErgoDox EZ - 0x3/0x1/0x1 - 0x41130000 - ICDeviceDescriptionUpdated)
default	11:18:34.715729 +0100	icdd	#ICDebug - 509:{ICDDMessageCenter.m} (-Rem ErgoDox EZ - 0x3/0x0/0x0 - 0x41130000 - ICDeviceDescriptionUpdated)
default	11:18:34.715924 +0100	icdd	#ICDebug - 509:{ICDDMessageCenter.m} (-Rem ErgoDox EZ - 0x3/0x1/0x2 - 0x41130000 - ICDeviceDescriptionUpdated)
default	11:18:34.716103 +0100	icdd	#ICDebug - 509:{ICDDMessageCenter.m} (-Rem ErgoDox EZ - 0x3/0x0/0x0 - 0x41130000 - ICDeviceDescriptionUpdated)
default	11:18:34.716287 +0100	icdd	#ICDebug - 509:{ICDDMessageCenter.m} (-Rem ErgoDox EZ - 0x1/0x1/0x0 - 0x41130000 - ICDeviceDescriptionUpdated)
default	11:18:34.716535 +0100	icdd	#ICDebug - 509:{ICDDMessageCenter.m} (-Rem ErgoDox EZ - 0x1/0x3/0x0 - 0x41130000 - ICDeviceDescriptionUpdated)
```

fredizzimo, 2017-01-16:
Based on the information my best guess is the midi functionality of the keyboard. Perhaps the new sysex API support.

Does the keymap set `MIDI_ENABLE`, or `API_SYSEX_ENABLE` to yes? The default Ergodox Ez should set both of those to no. If yes, could you try with it disabled?

I also noticed that @ezuk has made a mistake and by old habit put some of the options in `Makefile` instead of `rules.mk`, which most likely makes them have no effect. But that shouldn't matter in this case since the default should be off anyway. We should probably go through all the makefiles and add a comment to remind that the correct place to put the options is the rules.mk file.

szymonkaliski, 2017-01-17:
@fredizzimo thanks so much, I thought `MIDI_ENABLE` was disabled by default and I didn't check, my bad. Compiled without it and now I can use Ableton again :+1:

Shouldn't MIDI output work with Ableton though?

fredizzimo, 2017-01-17:
I also thought that it was disabled by default. At least it seems like it should be according to the makefiles. Was it enabled by default for the keymap you are using?

Anyway, yes. MIDI should work with Ableton. From the behaviour you described I'm suspecting that the keyboard crashes and goes into bootloader mode. That's probably caused by some midi command that the keyboard doesn't handle. Unfortunately I can't help you very much, since I don't have the programs in question. However I can give some pointers for either you or someone else that's going to debug the issue.

The easiest way to debug this would be with a hardware debugger. But that's not a normal thing to own, and you also need access to Ableton.

I'm not sure if it's possible to trap hard faults and send the callstack over the USB, but that could be another way to debug it.

The third way, is probably what has to be done, selectively turn of midi functionality until it starts working. Perhaps add some additional checks, and debug prints until the problem is found.

Since this is reported just now, I suspect it has to do with the API support that was recently added.

NoahAndrews, 2017-01-17:
I believe the new API, which is enabled by default, also enables MIDI support, because it currently uses MIDI to communicate.

fredizzimo, 2017-01-17:
Yes but the Ergodox Ez seems to at least try to disable it

https://github.com/jackhumbert/qmk_firmware/blob/master/keyboards/ergodox/ez/rules.mk

I also don't see where it's enabled by default unless the keyboard enables it. But I might be missing something.

skullydazed, 2017-06-27:
Is this still an ongoing issue we need to track, or can we close it now?

fredizzimo, 2017-06-27:
I think the the midi support on the Infinity Ergodox, and probably all other ChibiOS keyboards might still be broken, but a separate issue should probably be open for that.

drashna, 2018-10-21:
Sounds like it's safe to close this?

closed 2018-10-21 by drashna