qmk/qmk_firmware #1017: Linking error when building Let's Split

nicinabox, 2017-01-19:
I'm trying to put together a new keymap for the Let's Split based on my existing Planck keymap. It nearly builds but I'm getting hung up on this linking error and I'm not really sure where to go from here.

There are some oddities about how the `config.h` files are defined for this keyboard, but I think I've worked through most of that following the advice in #1008.

Any pointers would be greatly appreciated.

[Code here.](https://github.com/nicinabox/qmk_firmware/blob/master/keyboards/lets_split/keymaps/nic)

Relevant output:

```bash
~/code/qmk_firmware/keyboards/lets_split master
❯ make rev2-nic-dfu
Making lets_split/rev2 with keymap nic and target dfu

avr-gcc (GCC) 4.9.3
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: ./tmk_core/common/command.c                                                              [OK]
Linking: .build/lets_split_rev2_nic.elf                                                             [ERRORS]
 |
 | .build/obj_lets_split_rev2_nic/matrix.o: In function `serial_transaction':
 | /Users/nic/code/qmk_firmware/keyboards/lets_split/matrix.c:190: undefined reference to `serial_update_buffers'
 | /Users/nic/code/qmk_firmware/keyboards/lets_split/matrix.c:195: undefined reference to `serial_slave_buffer'
 | /Users/nic/code/qmk_firmware/keyboards/lets_split/matrix.c:195: undefined reference to `serial_slave_buffer'
 | .build/obj_lets_split_rev2_nic/matrix.o: In function `matrix_slave_scan':
 | /Users/nic/code/qmk_firmware/keyboards/lets_split/matrix.c:247: undefined reference to `serial_slave_buffer'
 | /Users/nic/code/qmk_firmware/keyboards/lets_split/matrix.c:247: undefined reference to `serial_slave_buffer'
 | .build/obj_lets_split_rev2_nic/split_util.o: In function `keyboard_master_setup':
 | /Users/nic/code/qmk_firmware/keyboards/lets_split/split_util.c:36: undefined reference to `serial_master_init'
 | .build/obj_lets_split_rev2_nic/split_util.o: In function `keyboard_slave_setup':
 | /Users/nic/code/qmk_firmware/keyboards/lets_split/split_util.c:44: undefined reference to `serial_slave_init'
 | collect2: error: ld returned 1 exit status
 |
make[1]: *** [.build/lets_split_rev2_nic.elf] Error 1
make: *** [rev2-nic-dfu] Error 1
Make finished with errors
```

nicinabox, 2017-01-19:
Looks like I needed to explicitly add `#define USE_SERIAL` to my `config.h` and it compiled successfully!

closed 2017-01-19 by nicinabox

ehrenmurdick, 2021-11-27:
@nicinabox Thank you for leaving the solution for the future!