qmk/qmk_firmware #1025: Adding support for Red Scarf II+

apierz, 2017-01-22:
I'd like to add support for the Red Scarf II+ keyboard. It currently uses TMK, but I prefer QMK and I'd like to use that instead. I think this is possible as the keyboard uses a ATmega32u4.

What are the necessary steps to get QMK to flash on a board that is not currently supported? So far I have gotten as far as trying to define the row and column pins but I don't know how to get that info. Any help anyone can give me would be greatly appreciated.

TerryMathews, 2017-01-23:
From https://github.com/kairyu/tmk_keyboard_custom/blob/master/keyboard/RedScarfII/matrix.c

'</* Column pin configuration
 * col: 0   1   2   3   4   5   6   7   8   9   10  11  12  13  14  15  16  17
 * pin: F4  F1  F0  B3  D0  D1  D4  D5  D6  D7  F7  F6  D2  D3  B6  B5  B4  F5
 */>'

'</* Row pin configuration
 * row:     0   1   2   3   4   5   x
 * pin: B2  0   1   0   1   0   1   1
 *      B1  0   0   1   1   0   0   1
 *      B0  0   0   0   0   1   1   1
 */>'

apierz, 2017-01-23:
Thanks for pointing me to that. In the other config.hs I've looked at I haven't seen a pin configuration like:
~~~~
    row: 0 1 2 3 4 5 x
    pin: B2 0 1 0 1 0 1 1
    B1 0 0 1 1 0 0 1
    B0 0 0 0 0 1 1 1
~~~~

How can I convert that to a set of set pins?

TerryMathews, 2017-01-23:
You can't. Wiring the matrix the way that they did allows them to compress what would've taken 6 pins (rows 0-5) and run it on 3 pins.

You're going to have to design a custom matrix.c to make it work. You can rework what was in the source I linked to you but the flags to tell QMK to use a custom matrix.c might be different - reference Let's Split. I know it has a custom matrix.c as well.

Considering TMK works, is there something specific that QMK does and TMK doesn't?

apierz, 2017-01-23:
Mostly little things like space cadet and tap dance that I have gotten used to on my Planck.

jackhumbert, 2017-02-07:
If you wanna copy over the matrix.c from TMK, you'd really only need to add `matrix_init_quantum();` and `matrix_scan_quantum();` to get all of the features. You can [reference the EZ](https://github.com/qmk/qmk_firmware/tree/master/keyboards/ergodox/ez) on how to use the custom matrix.

coryshaw1, 2017-04-11:
Looking to add support for Red Scarf Ver III (RS96). I'm not entirely sure where to start, guessing [this matrix](https://github.com/kairyu/tmk_keyboard_custom/blob/master/keyboard/RedScarfIII/matrix.c) will need to be ported?

jceaser, 2017-05-15:
any progress here, is there a branch or a fork where people are doing this? I just got a board and will be soldering it up soon, so I can compile something for people.

jackhumbert, 2017-05-15:
@coryshaw1 yeap! That's the correct file :)

jceaser, 2017-05-18:
@coryshaw1 have you started anything yet? I've been looking at the same file

coryshaw1, 2017-05-18:
@jceaser I have not personally. Just haven't had time yet

drashna, 2018-10-21:
Was this every worked on? 

@mechmerlin want to take a crack at this?

Maartenwut, 2019-05-24:
Got a keymap working, just need to figure out the LED situation.

drashna, 2019-06-05:
well, for the LEDs:
https://github.com/kairyu/tmk_keyboard_custom/blob/master/keyboard/RedScarfII/yc059.c

And I've harassed @mechmerlin about this as well.  So ne should be able to get rgb working, assuming he remembers. :)

drashna, 2019-11-21:
This has definitely been added. Closing

closed 2019-11-21 by drashna

asgeirtj, 2024-11-06:
> Got a keymap working, just need to figure out the LED situation.

Care to share how you did it? Am stranded with a red scarf iii rs96 which cant be flashed anymore because the tools are literally off the internet now because of chrome app depreciation. I have no idea how to do it now, I have my keyobard layout editor setup i want to flash, my use case should be pretty easy since i dont have any leds and only use one layer, don't use any fn. I've spent like 24 hours trying to hack this with some tmk repos and help of llm's but in the end I have no Idea what I'm doing. I already ruined 2x other rs96 boards. After all the google and llm using in the world to try and fix this i'm completely stuck on this. I'll pay you if u want.