qmk/qmk_firmware #1028: Which is the Teensy Linux page?

Rovanion, 2017-01-23:
In the [instructions for the ergodox ez](https://github.com/jackhumbert/qmk_firmware/tree/master/keyboards/ergodox#ergodox-ez) it's mentioned that some instructions for flashing on Linux are on the Teensy Linux page, but no link is provided. I can't seem to find the page so if you could point me in the right direction that would be very sweet.

What I'm really after is how to flash using `teensy_loader_cli`, so if you happen to have instructions on how to use that I'd be doubly happy.

SjB, 2017-01-24:
# make sure you have all the dependencies.
make ergodox-ez-default
sudo teensy_loader_cli --mcu=atmega32u4 -w ergodox_ez_default.hex

you can get the teensy_loader_cli from https://www.pjrc.com/teensy/loader_cli.html

Rovanion, 2017-01-24:
Thank you!

I'll cook up a PR which adds this to the instructions I linked later in the day.

closed 2017-01-24 by Rovanion