qmk/qmk_firmware #1050: Compiling error, infinite loop

cortesejea, 2017-01-29:
When compiling my files, enters a loop "make" command

`
C:\Users\Cortes\Desktop\qmk_firmware-master\keyboards\Shiro>make
Making Shiro with keymap default

Making Shiro with keymap default

Making Shiro with keymap default
...
`

Using -d this is what you see

`
C:\Users\Cortes\Desktop\qmk_firmware-master\keyboards\Shiro>make -d
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-pc-mingw32
find_and_set_shell path search set default_shell = C:/WinAVR-20100110/utils/bin/sh.exe
Reading makefiles...
Reading makefile `makefile'...
Reading makefile `../../Makefile' (search path) (no ~ expansion)...
Reading makefile `../..//message.mk' (search path) (no ~ expansion)...
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "awk --version 2>/dev/null",...)
Main thread handle = 0x00000130
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "git describe --abbrev=6 --dirty --always --tags 2>/dev/null || date +\"%Y-%m-%d-%H:%M:%S\"",...)
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "date +\"%Y-%m-%d-%H:%M:%S\"",...)
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "echo '#define QMK_VERSION \"2017-01-29-18:22:05\"' > ../..//quantum/version.h",...)
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "echo '#define QMK_BUILDDATE \"2017-01-29-18:22:05\"' >> ../..//quantum/version.h",...)
Reading makefile `../..//testlist.mk' (search path) (no ~ expansion)...
Reading makefile `../..//quantum/serial_link/tests/testlist.mk' (search path) (no ~ expansion)...
Updating makefiles....
 Considering target file `../..//quantum/serial_link/tests/testlist.mk'.
  Looking for an implicit rule for `../..//quantum/serial_link/tests/testlist.mk'.
  Trying pattern rule with stem `testlist.mk'.
  Found an implicit rule for `../..//quantum/serial_link/tests/testlist.mk'.
  Finished prerequisites of target file `../..//quantum/serial_link/tests/testlist.mk'.
 No need to remake target `../..//quantum/serial_link/tests/testlist.mk'.
 Considering target file `../..//testlist.mk'.
  Looking for an implicit rule for `../..//testlist.mk'.
  Trying pattern rule with stem `testlist.mk'.
  Found an implicit rule for `../..//testlist.mk'.
  Finished prerequisites of target file `../..//testlist.mk'.
 No need to remake target `../..//testlist.mk'.
 Considering target file `../..//message.mk'.
  Looking for an implicit rule for `../..//message.mk'.
  Trying pattern rule with stem `message.mk'.
  Found an implicit rule for `../..//message.mk'.
  Finished prerequisites of target file `../..//message.mk'.
 No need to remake target `../..//message.mk'.
 Considering target file `../../Makefile'.
  Looking for an implicit rule for `../../Makefile'.
  Trying pattern rule with stem `Makefile'.
  Found an implicit rule for `../../Makefile'.
  Finished prerequisites of target file `../../Makefile'.
 No need to remake target `../../Makefile'.
 Considering target file `makefile'.
  Looking for an implicit rule for `makefile'.
  Trying pattern rule with stem `makefile'.
  Found an implicit rule for `makefile'.
  Finished prerequisites of target file `makefile'.
 No need to remake target `makefile'.
Updating goal targets....
Considering target file `Shiro-allsp-allkm'.
 File `Shiro-allsp-allkm' does not exist.
 Looking for an implicit rule for `Shiro-allsp-allkm'.
 Trying pattern rule with stem `Shiro-allsp-allkm'.
 Found an implicit rule for `Shiro-allsp-allkm'.
 Finished prerequisites of target file `Shiro-allsp-allkm'.
Must remake target `Shiro-allsp-allkm'.
Reading makefile `../..//keyboards/Shiro/Makefile' (search path) (no ~ expansion)...
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "# Check if we have the CMP tool installed",...)
Putting child 0x02935d40 (Shiro-allsp-allkm) PID 43159384 on the chain.
Live child 0x02935d40 (Shiro-allsp-allkm) PID 43159384
Reaping winning child 0x02935d40 PID 43159384
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "cmp ../..//Makefile ../..//Makefile >/dev/null 2>&1; if [ $? -gt 0 ]; then printf \"\033[31;01mError:\033[0m\033[1m cmp command not found, please install diffutils\n\033[0m\"; exit 1; fi;",...)
Live child 0x02935d40 (Shiro-allsp-allkm) PID 43161224
Reaping winning child 0x02935d40 PID 43161224
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "# Check if the submodules are dirty, and display a warning if they are",...)
Live child 0x02935d40 (Shiro-allsp-allkm) PID 43161464
Reaping winning child 0x02935d40 PID 43161464
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "git submodule status --recursive 2>/dev/null | \
while IFS= read -r x; do \
        case \"$x\" in \
                \ *) ;; \
                *) printf \"\033[33;01mWARNING:\033[0m\n Some git sub-modules are out of date or modified, please consider runnning:\033[1m\n git submodule sync --recursive\n git submodule update --init --recursive\033[0m\n\n You can ignore this warning if you are not compiling any ChibiOS keyboards,\n or if you have modified the ChibiOS libraries yourself. \n\n\";break;; \
        esac \
done",...)
Live child 0x02935d40 (Shiro-allsp-allkm) PID 43160584
Reaping winning child 0x02935d40 PID 43160584
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "rm -f ../..//.build/error_occured > /dev/null 2>&1",...)
Live child 0x02935d40 (Shiro-allsp-allkm) PID 43160264
Reaping winning child 0x02935d40 PID 43160264
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "# Run all the commands in the same shell, notice the + at the first line",...)
Live child 0x02935d40 (Shiro-allsp-allkm) PID 43161224
Reaping winning child 0x02935d40 PID 43161224
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "# it has to be there to allow parallel execution of the submake",...)
Live child 0x02935d40 (Shiro-allsp-allkm) PID 43161224
Reaping winning child 0x02935d40 PID 43161224
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "# This always tries to compile everything, even if error occurs in the middle",...)
Live child 0x02935d40 (Shiro-allsp-allkm) PID 43160264
Reaping winning child 0x02935d40 PID 43160264
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "# But we return the error code at the end, to trigger travis failures",...)
Live child 0x02935d40 (Shiro-allsp-allkm) PID 43160264
Reaping winning child 0x02935d40 PID 43160264
CreateProcess(C:\WinAVR-20100110\utils\bin\sh.exe,C:/WinAVR-20100110/utils/bin/sh.exe -c "error_occured=0; printf \"Making \033[1mShiro\033[0m with keymap \033[1mdefault\033[0m\n\n\"; make -r -R -C ../../ -f build_keyboard.mk  KEYBOARD=Shiro SUBPROJECT= KEYMAP=default VERBOSE= COLOR=true SILENT=false; if [ $? -gt 0 ]; then error_occured=1; fi; if [ $error_occured -gt 0 ]; then echo $error_occured > ../..//.build/error_occured; fi;",...)
Live child 0x02935d40 (Shiro-allsp-allkm) PID 43160264
`

Here I leave the code of my keyboard incomplete, I will have to ask for help later if I can not use the backlight correctly.

https://www.dropbox.com/s/79r4943ufopp4d5/Shiro.rar?dl=0

fredizzimo, 2017-01-29:
The debug output only shows one iteration, so I don't see anything wrong there.

However, you have an old winavr version. So, I suggest you uninstall that and install the one linked in the instructions on the wiki instead.

cortesejea, 2017-01-30:
Okay, already solved the problem.
I have used the method of bash in windows 10 and it works fine, I have also solved a problem that I had wrong in the code.

I will create a new thread to know how to do the backlight.

closed 2017-01-30 by cortesejea