qmk/qmk_firmware #1051: Backlighting with QMK

cortesejea, 2017-01-30:
Hello, I have a problem creating the backlight configuration.

I have the keyboard with a led for each switch, in total 93 white leds connected to the Pin B5 (not RGB) (TKL + 5 keys)

I would like to have 3 keys to control the LEDs using PWM, 
ON / OFF
Increase
Decrease

This is the configuration of my files. (RAR -> https://www.dropbox.com/s/jo00qwws0xrnlkr/shiro.rar?dl=0 )

I'm sure there is useless code for my keyboard, since I use Cluepad as an example.
I had not used QMK, but had TMK for my previous custom keyboard

I press the keys I assigned FN1 FN2 FN3 and there is no answer.

# config.h

```
#ifndef CONFIG_H
#define CONFIG_H

#include "config_common.h"

/* USB Device descriptor parameter */
#define VENDOR_ID       0xC1ED
#define PRODUCT_ID      0x2312
#define DEVICE_VER      0x0001
#define MANUFACTURER    Cortes
#define PRODUCT         Shiroboard
#define DESCRIPTION     QMK keyboard firmware for Shiroboard

/* key matrix size */
#define MATRIX_ROWS 6
#define MATRIX_COLS 17

#define MATRIX_ROW_PINS { F7, F6, F5, F4, F1, F0 }
#define MATRIX_COL_PINS { B4, E6, D7, D6, D4, D5, C7, C6, D3, D2, D1, D0, B7, B3, B2, B1, B0 }
#define UNUSED_PINS

#define DIODE_DIRECTION COL2ROW

#define DEBOUNCING_DELAY 5

#define BACKLIGHT_LEVELS 3
#define BACKLIGHT_PIN B5

#define LOCKING_SUPPORT_ENABLE

#define LOCKING_RESYNC_ENABLE

#define IS_COMMAND() ( \
    keyboard_report->mods == (MOD_BIT(KC_LSHIFT) | MOD_BIT(KC_RSHIFT)) \
)

#define ALWAYS_DEBUG

#endif
```

# rules.mk
```
MCU = atmega32u4

F_CPU = 16000000

ARCH = AVR8

F_USB = $(F_CPU)

OPT_DEFS += -DINTERRUPT_CONTROL_ENDPOINT

OPT_DEFS += -DBOOTLOADER_SIZE=4096

BOOTMAGIC_ENABLE ?= yes
# MOUSEKEY_ENABLE ?= no
# EXTRAKEY_ENABLE ?= no
# CONSOLE_ENABLE ?= no
# COMMAND_ENABLE ?= no
NKRO_ENABLE ?= no
BACKLIGHT_ENABLE ?= yes
#RGBLIGHT_ENABLE ?= no
# MIDI_ENABLE ?= no
# UNICODE_ENABLE ?= no
# BLUETOOTH_ENABLE ?= no
```

# shiro.c
```
#include "shiro.h"

int pwm_level;

void matrix_init_kb(void) {
    // put your keyboard start-up code here
    // runs once when the firmware starts up
    matrix_init_user();

    // JTAG disable for PORT F. write JTD bit twice within four cycles.
    MCUCR |= (1<<JTD);
    MCUCR |= (1<<JTD);
};

void led_set_kb(uint8_t usb_led)
{
    print("led_set\n");
}

void backlight_init_ports(void)
{
    // Set C7 to output
    DDRB |= (1<<5);

    // Initialize the timer
    TC4H = 0x03;
    OCR4B = 0xFF;
    TCCR4A = 0b10000010;
    TCCR4B = 0b00000001;
}

void backlight_set(uint8_t level)
{
    // Determine the PWM level
    switch (level)
    {
        case 0:
            // 33%
            pwm_level = 0x54;
            break;
        case 1:
            // 66%
            pwm_level = 0xA8;
            break;
        case 2:
            // 100%
            pwm_level = 0xFF;
            break;
        case 3:
            // 0%
            pwm_level = 0x00;
            break;
        default:
            xprintf("Unknown level: %d\n", level);
    }

    // Write the PWM level to the timer
    TC4H = pwm_level >> 8;
    OCR4B = 0xFF & pwm_level;
}
```
# shiro.h
```
#ifndef SHIRO_H
#define SHIRO_H

#include "quantum.h"

#define KEYMAP( \
    K0A,      K0C, K0D, K0E, K0F, K0G, K0H, K0I, K0J, K0K, K0L, K0M, K0N,       K0O, K0P, K0Q, \
    K1A, K1B, K1C, K1D, K1E, K1F, K1G, K1H, K1I, K1J, K1K, K1L, K1M, K1N,       K1O, K1P, K1Q, \
    K2A, K2B, K2C, K2D, K2E, K2F, K2G, K2H, K2I, K2J, K2K, K2L, K2M, K2N,       K2O, K2P, K2Q, \
    K3A, K3B, K3C, K3D, K3E, K3F, K3G, K3H, K3I, K3J, K3K, K3L, K3M,            K3O, K3P, K3Q, \
    K4A, K4B, K4C, K4D, K4E, K4F, K4G, K4H, K4I, K4J, K4K, K4L, K4M,            K4O, K4P, K4Q, \
	K5A, K5B, K5C,                K5G,                K5K, K5L, K5M, K5N,       K5O, K5P, K5Q  \
) { \
    { K0A,  KC_NO,K0C,  K0D,  K0E,  K0F,  K0G,  K0H,  K0I,  K0J,  K0K,  K0L,  K0M,  K0N,          K0O,  K0P,  K0Q }, \
    { K1A,  K1B,  K1C,  K1D,  K1E,  K1F,  K1G,  K1H,  K1I,  K1J,  K1K,  K1L,  K1M,  K1N,          K1O,  K1P,  K1Q }, \
    { K2A,  K2B,  K2C,  K2D,  K2E,  K2F,  K2G,  K2H,  K2I,  K2J,  K2K,  K2L,  K2M,  K2N,          K2O,  K2P,  K2Q }, \
    { K3A,  K3B,  K3C,  K3D,  K3E,  K3F,  K3G,  K3H,  K3I,  K3J,  K3K,  K3L,  K3M,  KC_NO,        K3O,  K3P,  K3Q }, \
    { K4A,  K4B,  K4C,  K4D,  K4E,  K4F,  K4G,  K4H,  K4I,  K4J,  K4K,  K4L,  K4M,  KC_NO,        K4O,  K4P,  K4Q }, \
    { K5A,  K5B,  K5C,  KC_NO,KC_NO,KC_NO,K5G,  KC_NO,KC_NO,KC_NO,K5K,  K5L,  K5M,  K5N,          K5O,  K5P,  K5Q }  \
}

#endif
```
# keymap.c
```
#include "shiro.h"
#include "backlight.h"

#define _BL 0
#define _FL 1

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
[_BL] = KEYMAP(
  KC_ESC,            KC_F1,  KC_F2,  KC_F3,  KC_F4,  KC_F5,  KC_F6,  KC_F7,  KC_F8,  KC_F9,  KC_F10,  KC_F11,  KC_F12,     KC_PSCR,KC_SLCK,KC_PAUS, \
  KC_GRV,   KC_1,    KC_2,   KC_3,   KC_4,   KC_5,   KC_6,   KC_7,   KC_8,   KC_9,   KC_0,   KC_MINS, KC_EQL,  KC_BSPC,    KC_INS, KC_HOME,KC_PGUP, \
  KC_TAB,   KC_Q,    KC_W,   KC_E,   KC_R,   KC_T,   KC_Y,   KC_U,   KC_I,   KC_O,   KC_P,   KC_LBRC, KC_RBRC, KC_ENT,     KC_DEL, KC_END, KC_PGDN, \
  KC_CAPS,  KC_A,    KC_S,   KC_D,   KC_F,   KC_G,   KC_H,   KC_J,   KC_K,   KC_L,   KC_SCLN,KC_QUOT, KC_NUHS,             KC_FN2, KC_FN1, KC_TRNS, \
  KC_LSFT,  KC_NUBS, KC_Z,   KC_X,   KC_C,   KC_V,   KC_B,   KC_N,   KC_M,   KC_COMM,KC_DOT, KC_SLSH, KC_RSFT,             KC_FN3, KC_UP,  KC_TRNS, \
  KC_LCTL,  KC_LGUI, KC_LALT,                        KC_SPC,                         KC_RALT,KC_FN0,  KC_TRNS, KC_RCTL,    KC_LEFT,KC_DOWN,KC_RGHT), 
  
[_FL] = KEYMAP(
  KC_ESC,            KC_F1,  KC_F2,  KC_F3,  KC_F4,  KC_F5,  KC_F6,  KC_F7,  KC_F8,  KC_F9,  KC_F10,  KC_F11,  KC_F12,     KC_PSCR,KC_SLCK,KC_PAUS, \
  KC_GRV,   KC_1,    KC_2,   KC_3,   KC_4,   KC_5,   KC_6,   KC_7,   KC_8,   KC_9,   KC_0,   KC_MINS, KC_EQL,  KC_BSPC,    KC_INS, KC_HOME,KC_PGUP, \
  KC_TAB,   KC_Q,    KC_W,   KC_E,   KC_R,   KC_T,   KC_Y,   KC_U,   KC_I,   KC_O,   KC_P,   KC_LBRC, KC_RBRC, KC_ENT,     KC_DEL, KC_END, KC_PGDN, \
  KC_CAPS,  KC_A,    KC_S,   KC_D,   KC_F,   KC_G,   KC_H,   KC_J,   KC_K,   KC_L,   KC_SCLN,KC_QUOT, KC_NUHS,             KC_FN2, KC_FN1, KC_TRNS, \
  KC_LSFT,  KC_NUBS, KC_Z,   KC_X,   KC_C,   KC_V,   KC_B,   KC_N,   KC_M,   KC_COMM,KC_DOT, KC_SLSH, KC_RSFT,             KC_FN3, KC_UP,  KC_TRNS, \
  KC_LCTL,  KC_LGUI, KC_LALT,                        KC_SPC,                         KC_RALT,KC_FN0,  KC_TRNS, KC_RCTL,    KC_LEFT,KC_DOWN,KC_RGHT), 
};

/*enum function_id {
};*/

const uint16_t PROGMEM fn_actions[] = {
    [0] = ACTION_LAYER_MOMENTARY(1),  // to Fn overlay
	[1] = ACTION_BACKLIGHT_TOGGLE(),
	[2] = ACTION_BACKLIGHT_INCREASE(),
	[3] = ACTION_BACKLIGHT_DECREASE(),
};
```

drashna, 2018-03-25:
Is this still an issue?

cortesejea, 2018-03-25:
I did not get it to work, so I gave up.

drashna, 2018-03-25:
I'm sorry to hear that! 
In case somebody has some insight, going to leave this open for now.

closed 2019-03-29 by drashna