qmk/qmk_firmware #1060: Remapping Caps Lock to Ctrl when holding and Escape when tapping.

zkanda, 2017-02-01:
Hi guys.

I'm looking for a keyboard and as such I'm trying to find the right firmware for a keyboard.

I have a question for keyboard mapping. I'm a vim user so it's probably a bit specific but in particular, here's what I'm hoping to achieve.

When holding caps lock key, it would become control.
When tapping caps lock key, it would become escape.

I'm currently using a software to do it but it would be good to be able to make it in the hardware.
You can find more information here: http://www.economyofeffort.com/2014/08/11/beyond-ctrl-remap-make-that-caps-lock-key-useful/

So is this possible for qmk_firmware?

skullydazed, 2017-02-01:
Yes, you would use MT() as described here:

https://github.com/qmk/qmk_firmware/wiki#fun-with-modifier-keys

In your case you would want this in that key location: MT(MOD_LCTL, KC_ESC)

zkanda, 2017-02-02:
Thank you so much, I'll closed this since I will need it when I get the new keyboard.

closed 2017-02-02 by zkanda