qmk/qmk_firmware #1061: Backlighting not working with Minivan/TV44

robotmaxtron, 2017-02-01:
As per @evangs the backlight pin for TV44 is B2, not B7 as it is committed here: https://github.com/qmk/qmk_firmware/blob/master/keyboards/tv44/config.h

Changing this pin to B2 causes a failure.

```
Compiling: keyboards/tv44/tv44.c                                                                    [OK]
Compiling: keyboards/tv44/keymaps/robotmaxtron/keymap.c                                             [OK]
Compiling: quantum/quantum.c                                                                       quantum/quantum.c:528:4: error: #error "Backlight pin not supported - use B5, B6, or B7"
 #  error "Backlight pin not supported - use B5, B6, or B7"
    ^~~~~
In file included from /usr/avr/include/avr/io.h:99:0,
                 from /usr/avr/include/avr/pgmspace.h:90,
                 from quantum/quantum.h:5,
                 from quantum/quantum.c:1:
quantum/quantum.c: In function 'backlight_init_ports':
quantum/quantum.c:554:16: error: 'COM1x1' undeclared (first use in this function)
   TCCR1A = _BV(COM1x1) | _BV(WGM11); // = 0b00001010;
                ^
quantum/quantum.c:554:16: note: each undeclared identifier is reported only once for each function it appears in
quantum/quantum.c: In function 'backlight_set':
quantum/quantum.c:572:21: error: 'COM1x1' undeclared (first use in this function)
     TCCR1A &= ~(_BV(COM1x1));
                     ^
quantum/quantum.c:573:5: error: 'OCR1x' undeclared (first use in this function)
     OCR1x = 0x0;
     ^~~~~
quantum/quantum.c: In function '__vector_17':
quantum/quantum.c:789:5: error: 'OCR1x' undeclared (first use in this function)
     OCR1x = (uint16_t)(((uint16_t)pgm_read_byte(&breathing_table[local_index]) * 257)) >> breath_intensity;
     ^~~~~
 [ERRORS]
 | 
 | 
 | 
tmk_core/rules.mk:352: recipe for target '.build/obj_tv44_robotmaxtron/quantum/quantum.o' failed
make[1]: *** [.build/obj_tv44_robotmaxtron/quantum/quantum.o] Error 1
Make finished with errors
../../Makefile:489: recipe for target 'robotmaxtron' failed
make: *** [robotmaxtron] Error 1
```

shelaf, 2017-02-02:
The backlight funtion of QMK is adjusting LED brightness by PWM(Pulse Width Modulation).
On ATmega32u4, PWM is suppoted by PB5, PB6, PB7, PC6, PC7, PD0. ("P" means "Port")
But, QMK firmware supports only PB5, PB6, PB7.
https://www.arduino.cc/en/Hacking/PinMapping32u4

On TV44 keyboard, as mentioned below, backlight is toggled by the caps lock state and be on or off only.
So, i think, you do not need to enable QMK backlight function.
https://geekhack.org/index.php?topic=80988.150

jackhumbert, 2017-02-02:
In addition to the PWM ports mentioned, I would like to eventually support regular IO pins that would just have a boolean status (they'd be forced to have just 1 backlight level).

jackhumbert, 2017-02-07:
@robotmaxtron can you checkout #1073, and try to use the B2 pin? I'm not sure what `BACKLIGHT_ON_STATE` should be, but you should be able to try both safely.

robotmaxtron, 2017-02-08:
@jackhumbert Sure, I'll give it a shot and report back.

robotmaxtron, 2017-02-09:
@jackhumbert It worked! In addition to your change from #1073 , my config changes shown here: https://github.com/robotmaxtron/qmk_firmware/commit/02caf7f8d98b998443adf07b4b97fd915d9d8538 allowed me to set 1 to enable backlighting and 0 to disable backlighting.

Please let me know if I can help test anything else and thank you for getting this added!

jackhumbert, 2017-02-09:
Awesome! The `BACKLIGHT_ON_STATE` should reverse the logic - have you tried turning it on/off via the keycodes? You should be able to step through the levels as well, but there will only be on/off right now - I'm looking into a way to make it dimmable via soft-PWM :)

robotmaxtron, 2017-02-09:
I'll give the levels and turning on/off via keycodes a shot as well.

robotmaxtron, 2017-02-10:
@jackhumbert Okay so using `ACTION_BACKLIGHT_TOGGLE()` within the keymapping works. Using `ACTION_BACKLIGHT_INCREASE()` and `ACTION_BACKLIGHT_DECREASE()` does appear to control the levels but it does not actually appear to change the brightness of the leds.

By this I mean if you decrease it enough, they turn off. If you toggle while off, they stay off. If you increase while "lowest", they turn on.

jackhumbert, 2017-02-10:
Perfect :) that's exactly what should happen haha. I'll see if I can mess around with the levels soon and tag you again if that's ok!

jackhumbert, 2017-02-12:
@robotmaxtron would you mind trying #1090? You shouldn't need to change anything from last time. It should allow you to have some different brightness levels though :)

ekx, 2017-10-12:
I've been trying to implement backlight for the winkeyless b.face keyboard, which uses the D4 pin for the backlight LEDs. When I hook that pin up as the num lock LED it works fine and I can toggle the backlight with the num lock key.
When I set "BACKLIGHT_ENABLE" to "yes" and define the "BACKLIGHT_PIN" the backlight will randomly be on or off (regardless whether "BACKLIGHT_ON_STATE" is 0 or 1) while the "BL_" keys seem to do nothing.
Code is available [here](https://github.com/ekx/qmk_firmware/tree/bface/keyboards/ps2avrGB)
Kinda stumped right now since on top of all else I can' seem to get "hid_listen" to connect to the board...

drashna, 2018-03-25:
Is this still an issue?