qmk/qmk_firmware #1064: Trembling Breathing Backlight Under Low Speed

SuJiKiNen, 2017-02-03:
this appears on my tada68,when default speed set at 3,the light effect is correct,when I decrease the speed,the LED lights change unsmoothly.
and when change backlight level at breathing mode,the LEDS will flash once,is this normal?

fredizzimo, 2017-03-26:
I'm not sure what kind of effect you are seeing. But if the problem is significantly worse with low brightness and improves when you increase the brightness it's probably the following.

Due to the limited number of steps (256), and the non-linearity of LEDs it's impossible to have smooth fading at low brightness at low speed without special tricks, something that QMK currently doesn't implement. What we need is [temporal dithering](https://en.wikipedia.org/wiki/Frame_rate_control). [Here's a video](https://www.youtube.com/watch?v=1-_JtRl2ks0) showing an effect without and with it.

The problem could of course be something else too, there has been some refactoring to the LED and backlight support recently.

fredizzimo, 2017-03-26:
[Here's a bit better description](https://github.com/FastLED/FastLED/wiki/FastLED-Temporal-Dithering) than the Wikipedia article.

skullydazed, 2017-10-22:
I believe that #1692 fixes this.

closed 2017-10-22 by skullydazed