qmk/qmk_firmware #1067: Diodes soldered wrong way; DIODE_DIRECTION ROW2COL shows no effect?

chief-wrench, 2017-02-05:
Hi folks,   
half way through handwiring something what I call the "Lets Split A Preonic" I found that I have soldered all the diodes the wrong way. Mock me.

The board works like expected if i interchange the row and column pins: The key rows and columns are rotated 90 degrees, but they do work.

"No problem," I thought, "I will just define DIODE_DIRECTION ROW2COL instead of COL2ROW." Only to find out it does not change anything.

My directory is a duplicate of lets_split and lives in keyboards like the original. I have added my additional row(s), added a pin to the pro micro pin definition for my additional row and that is about all my changes so far. I have not messed with any logic parts of QMK.

Mockery and help is welcome!

shelaf, 2017-02-06:
As you thought, just define DIODE_DIRECTION ROW2COL.
All theory of mechanical keyboard matrix is described in below link.Check your logic carefully.
http://blog.komar.be/how-to-make-a-keyboard-the-matrix/

chief-wrench, 2017-02-06:
Yes, I tried that already. But it does not change the behaviour. My guess is that my definition is overridden by some other definition that takes precedence.

jackhumbert, 2017-02-07:
Unfortunately just changing the diode direction isn't all that needs done in this case - the logic needs to be inverted, which isn't currently supported. If you wanted to use the ROW2COL setting, your diodes would need to be on the opposite side of the keyswitches in the matrix. I don't think there's a whole lot of interest to implement the logic inversion, but it would be nice to have in the future.

shensmobile, 2017-02-16:
Hi Chief, 

I am currently in the same situation as you.  I have a keyboard wired up where the diodes were easier to connect vertically so I did them like a 90 degree rotated Planck, and as suggested from the blog listed above: http://blog.komar.be/wp-content/uploads/2013/09/f.png

Neither ROW2COL nor COL2ROW work, and the description given in the file makes no sense because our Cathodes are pointing toward the column, but that will not allow the GND signal through.

Is there somewhere where this is explained more clearly in the documentation, or did I misinterpret the explanation of ROW2COL in the code comments?

jackhumbert, 2017-03-29:
@shensmobile can you point me to where you're seeing the description you mentioned? The cathodes should be pointing away from the switches using this logic.

shensmobile, 2017-03-29:
@jackhumbert this issue is specific to the Let's Split firmware.  I discovered that in the Let's Split, there is no reference to DIODE_DIRECTION in any file other than where you specify ROW2COL or COL2ROW.  I tried patching it in manually, and it works, but then I2C wouldn't work afterwards, so I gave up and just reversed my row and col definitions.

jackhumbert, 2017-03-29:
Ah, doy. Yeah, the custom Let's Split matrix doesn't support DIODE_DIRECTION, but could probably be adapted to! For now, the ROW2COL stuff should probably be removed from the config.h.

closed 2017-06-25 by skullydazed

Stormranger404, 2021-08-09:
I have an ergodox that i have hand wired, printed with a 3d printer, it was my first make project, and im having the same issue, where in the living duck is the logic defined anyways? if i could just get voltage on the opposite side of my diodes without the keyboard thinking im pressing literally every key.

I've even tried to invert my columns and my rows (i'm not attached to either word and ergodoxs are actually messed up in their logic anyways) and i get about as far as modifying the matrix and i get endless errors thrown at me. it's enough i almost never want to make anything ever again.

commit 62c77e7d (2017-06-25): Remove the DIODE_DIRECTION defines from lets_split so we can close #1067