qmk/qmk_firmware #1077: Ergodox Infinity LED matrix handling

yiancar, 2017-02-09:
Not an issue but rather a question. I was looking to in the Ergodox Infinity files to try to determine how you control the LED matrix (IS31FL3731C) where exactly is that located as I couldnt find it?

SjB, 2017-02-13:
I was told that the visualiser will be changing but for now you could look at my keymap at https://github.com/SjB/qmk_firmware/tree/sjb-keymaps/keyboards/ergodox/keymaps/sjb

fredizzimo, 2017-03-26:
I will soon enable the old visualizer from my TMK fork, which should have LED support. It's quite untested and possibly with some issue as reported [https://github.com/fredizzimo/infinity_ergodox/issues/5]

The reason why it wasn't done, was that I wanted to do a bigger [effect system](https://github.com/qmk/qmk_firmware/issues/717). However for many reasons it's taken a very long time, and it still doesn't seem to be very near. So better make something that works first.

fredizzimo, 2017-04-12:
The LCD and visualizer support is now merged. But the LEDs are currently disabled and there's no animation that would use them. I believe the underlying code works though.

As I don't have LEDs installed myself, it would be good if someone else test the code and complete the support by adding it to both the keyboard default visualizer and the simple_visualizer, in a suitable way. But first it's probably worth testing the test animation that is included in the `animations.c` file. 

Unfortunately there's no documentation for this yet, except for some code comments, the test animation and the two visualizers (`infinity/visualizer.c` and `infinity/simple_visualizer.h` ) included. But it shouldn't be too hard to figure out if you are an experienced enough coder. Some of the code comments might have been deleted because of restructuring, so it might be worthwhile to check the [old documented code](https://github.com/qmk/qmk_firmware/blob/a11053bda8f6263cf85c6127975cf3364f25083c/quantum/visualizer/example_integration/visualizer_user.c) as well 

One thing to note is that the LEDs are exposed as an uGFX display in gray color, so it supports controlling individual LEDs by plotting "pixel", but also more high level things like lines and circles, and even text.

fredizzimo, 2017-05-24:
Has anyone been able to test the LED support, does it work and can the issue be closed?

packman86, 2017-05-26:
I am also very interested in getting my backlighting to work on my Ergodox Infinity with QMK.

fredizzimo, 2017-05-26:
I have the emulator in good enough shape now, that I could implement some basic backlight support, but I can't say when it will be done.

That would be enabled by BACKLIGHT_ENABLE, and just lit up all the LEDs, allowing for the normal QMK brightness controls.

When the keyboard goes to sleep, I will just disable the backlight. 

If someone else is willing to work on that, please inform me before you start, so that I know.

fredizzimo, 2017-06-03:
I have now made some initial support in https://github.com/qmk/qmk_firmware/pull/1361. It's not tested yet, so testers are welcome to try it out.

fredizzimo, 2017-06-19:
Basic LED support have now been merged in, so closing this.

closed 2017-06-19 by fredizzimo