qmk/qmk_firmware #1080: EZ configurator keymap.c infinity make error

leshow, 2017-02-09:
I have an Ergodox Infinity, one of the newer ones with USB-C & LCD screen. I read your readme so I know layouts inter-operate between the infinity/ez.

The EZ online configurator (http://configure.ergodox-ez.com/keyboard_layouts/) is superior to the input club and massdrop one in the number of options it provides. I wanted to take the keymap generated from it and flash it to my infinity. I downloaded the source generated from the online configurator, made a new dir in `keymaps` called `minelayout`, and put the source in that dir as `keymap.c`.

I then ran `make infinity-minelayout` and got this error:

```bash
ergodox - master! ❯ make infinity-minelayout
Making ergodox/infinity with keymap minelayout

arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/ergodox/infinity/matrix.c                                                      [OK]
Compiling: keyboards/ergodox/infinity/led.c                                                         [OK]
Compiling: ./tmk_core/protocol/chibios/usb_main.c                                                   [OK]
Compiling: ./tmk_core/protocol/chibios/main.c                                                       [OK]
Compiling: keyboards/ergodox/ergodox.c                                                              [OK]
Compiling: keyboards/ergodox/keymaps/minelayout/keymap.c                                           keyboards/ergodox/keymaps/minelayout/keymap.c: In function 'process_record_user':
keyboards/ergodox/keymaps/minelayout/keymap.c:69:9: warning: implicit declaration of function 'rgblight_mode' [-Wimplicit-function-declaration]
         rgblight_mode(1);
         ^
 [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                 In file included from ./tmk_core/common/action_macro.h:20:0,
                 from ./tmk_core/common/action.h:25,
                 from quantum/keymap.h:23,
                 from quantum/keymap_common.c:18:
quantum/keymap_common.c: In function 'keymap_function_id_to_action':
quantum/keymap_common.c:179:34: warning: array subscript is above array bounds [-Warray-bounds]
  return pgm_read_word(&fn_actions[function_id]);
                                  ^
./tmk_core/common/progmem.h:9:46: note: in definition of macro 'pgm_read_word'
 #   define pgm_read_word(p)     *((uint16_t*)p)
                                              ^
 [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/process_keycode/process_leader.c                                                 [OK]
Compiling: keyboards/ergodox/infinity/infinity.c                                                    [OK]
Compiling: quantum/process_keycode/process_unicode.c                                                [OK]
Compiling: ./quantum/serial_link/protocol/transport.c                                               [OK]
Compiling: ./quantum/serial_link/protocol/triple_buffered_object.c                                  [OK]
Compiling: ./quantum/serial_link/protocol/frame_validator.c                                         [OK]
Compiling: ./quantum/serial_link/protocol/frame_router.c                                            [OK]
Compiling: ./quantum/serial_link/protocol/byte_stuffer.c                                            [OK]
Compiling: ./quantum/serial_link/system/serial_link.c                                               [OK]
Compiling: ./tmk_core/common/host.c                                                                 [OK]
Compiling: ./tmk_core/common/keyboard.c                                                             [OK]
Compiling: ./tmk_core/common/action.c                                                               [OK]
Compiling: ./tmk_core/common/action_tapping.c                                                       [OK]
Compiling: ./tmk_core/common/action_macro.c                                                         [OK]
Compiling: ./tmk_core/common/action_layer.c                                                         [OK]
Compiling: ./tmk_core/common/action_util.c                                                          [OK]
Compiling: ./tmk_core/common/print.c                                                                [OK]
Compiling: ./tmk_core/common/debug.c                                                                [OK]
Compiling: ./tmk_core/common/util.c                                                                 [OK]
Compiling: ./tmk_core/common/eeconfig.c                                                             [OK]
Compiling: ./tmk_core/common/chibios/suspend.c                                                      [OK]
Compiling: ./tmk_core/common/chibios/timer.c                                                        [OK]
Compiling: ./tmk_core/common/chibios/bootloader.c                                                   [OK]
Compiling: ./tmk_core/common/chibios/printf.c                                                       [OK]
Compiling: ./tmk_core/common/chibios/eeprom.c                                                       [OK]
Compiling: ./tmk_core/common/magic.c                                                                [OK]
Compiling: ./tmk_core/common/mousekey.c                                                             [OK]
Compiling: ./tmk_core/common/command.c                                                              [OK]
Compiling: ./tmk_core/common/chibios/sleep_led.c                                                    [OK]
Compiling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/crt1.c                                [OK]
Compiling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/vectors.c                             [OK]
[... 54 lines cut ...]
Compiling: lib/chibios-contrib/os/hal/ports/KINETIS/LLD/hal_ext_lld.c                               [OK]
Compiling: lib/chibios-contrib/os/hal/ports/KINETIS/LLD/hal_adc_lld.c                               [OK]
Compiling: lib/chibios-contrib/os/hal/ports/KINETIS/LLD/hal_gpt_lld.c                               [OK]
Compiling: lib/chibios-contrib/os/hal/ports/KINETIS/K20x/hal_pwm_lld.c                              [OK]
Compiling: lib/chibios-contrib/os/hal/ports/KINETIS/LLD/hal_st_lld.c                                [OK]
Compiling: lib/chibios-contrib/os/hal/ports/KINETIS/LLD/hal_usb_lld.c                               [OK]
Compiling: lib/chibios-contrib/os/hal/boards/PJRC_TEENSY_3_1/board.c                                [OK]
Compiling: lib/chibios/os/hal/lib/streams/chprintf.c                                                [OK]
Compiling: lib/chibios/os/hal/lib/streams/memstreams.c                                              [OK]
Compiling: lib/chibios/os/hal/lib/streams/nullstreams.c                                             [OK]
Assembling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/crt0_v7m.S                           [OK]
Assembling: lib/chibios/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v7m.S                        [OK]
Linking: .build/ergodox_infinity_minelayout.elf                                                     [ERRORS]
 | 
 | .build/obj_ergodox_infinity_minelayout/keyboards/ergodox/keymaps/minelayout/keymap.o: In function `process_record_user':
 | .build/obj_ergodox_infinity_minelayout/keyboards/ergodox/keymaps/minelayout/keymap.o: In function `process_record_user':
 | /home/evan/personal/config/qmk_firmware/keyboards/ergodox/keymaps/minelayout/keymap.c:69: undefined reference to `rgblight_mode'
 | collect2: error: ld returned 1 exit status
 | 
tmk_core/rules.mk:279: recipe for target '.build/ergodox_infinity_minelayout.elf' failed
make[1]: *** [.build/ergodox_infinity_minelayout.elf] Error 1
../../Makefile:489: recipe for target 'infinity-minelayout' failed
Make finished with errors
make: *** [infinity-minelayout] Error 1
```

leshow, 2017-02-09:
Looking at the default keymap.c vs the one generated by the EZ configurator
default/keymap.c:
```
    case RGB_SLD:
      if (record->event.pressed) {
        #ifdef RGBLIGHT_ENABLE
          rgblight_mode(1);
        #endif
      }
      return false;
```
from ez configurator
```
    case RGB_SLD:
      if (record->event.pressed) {
        rgblight_mode(1);
      }
      return false;
      break;
    
  }
```

jackhumbert, 2017-02-09:
As far as I know, the RGB stuff won't work on ARM, so adding the ifdef or removing the rgblight line should allow things to work :)

The blank function isn't necessary - it's just there in case you'd like to add something to it, but it doesn't hurt to have.

I don't think it's possible to brick ARM by flashing the wrong keymap (it looks like this one will work though), but I'm not the most experienced in ARM right now.

I don't know if QMK's setup to control the LCD on the infinity right now unfortunately!

leshow, 2017-02-09:
Thanks. So I was able to flash the firmware and everything appears to be working except for the LCD screens. There is no light coming from them and they aren't displaying anything.

r2d2rogers, 2017-02-09:
I have a branch on my fork of qmk that uses the visualizer files to change the background color and display the layer name on the LCD screen, based on @fredizzimo 's TMK fork. I haven't taken the time to do more than see what works.  I'm waiting for the new effects system he is working on to do more than that.

leshow, 2017-02-09:
@r2d2rogers awesome! I may try flashing yours then, which branch are the changes on? I don't want anything fancy, I'd just like to see the layer number and have the colour change.

Are you waiting for the 'effects system' to be integrated into QMK or in fredizzimo's TMK fork?

r2d2rogers, 2017-02-09:
Dual role is exactly what I jumped over here for myself.  It's the r2d2rogers branch of my fork of qmk_firmware.  I was waiting till I filled in a bit more of the spaceFN style layers before I submitted to a pull request.  for the dual role keys, see how I'm using what would be caps lock for a ctrl/esc, and the right hand thumb key that is space and momentary layer, for the spaceFN I was talking about.

leshow, 2017-02-09:
@r2d2rogers I have a keymap.c with my own layout an dual role keys set up already. Can I not just overwrite your keymap in `ergodox/keymaps/r2d2rogers` and then flash?

r2d2rogers, 2017-02-09:
That should work, sorry, didn't read up top and see that you had already gotten a keymap setup.

closed 2017-03-07 by leshow

fredizzimo, 2017-03-26:
Regarding the original bug. The plan is to have either empty implementation or some kind of emulation of the Ergodox features. So in this case you shouldn't need to  add `#ifdefs` around `rgblight_mode`. Many other functions are already implemented as dummies, but this function is not for some reason. Is there no keymap in the repository that actually uses it? Since in that case, it should have caused errors in our continous integration system as well.

In any case, I will add an empty implementation to make it work without changes.