qmk/qmk_firmware #1081: Is there interest in adding support for boards using GON NerD PCBs?

Mauin, 2017-02-09:
Over the last week I've worked on getting QMK running on my GON NerD based board. There is support for the nerd in TMK in the "orphan" directory as well as some forks which I've taken as a base to get QMK running on my board.

Due to the complicated nature of getting TMK/QMK to run on the NerD, having to manually flash the bootloader of the chip, is there any interest in adding support for it in this repo?

jackhumbert, 2017-02-09:
Yeah! That'd be awesome :)

Mauin, 2017-02-10:
Cool, I'll try to clean up my files over the weekend and then create a PR. Implementation is pretty minimal so far, I guess. Especially since the NerD supports multiple layouts and I only have my one layout to test it on.

PR #1088 (2017-02-11): add GON NerD
Resolves #1081 

This adds support for GON NerD PCB based keyboards. 

The keymaps should theoretically work for both the TKL and the 60% variant in all available layouts, however I only have a 60% variant in a Pok3r style layout, so that's the only one I could actually test and verify.
closed 2017-02-11 by jackhumbert