qmk/qmk_firmware #1086: Move documentation from wiki back to source?

priyadi, 2017-02-11:
Advantages:

* Code and documentation can be submitted in a single pull request. With wiki, documenting is a whole separate process.
* Contributors no longer need wait until a pull request is accepted, before starting to work on documentation.
* Jack and others with write access are able to ask for adequate documentation from incoming pull requests before accepting them.

Thoughts?

xtuc, 2017-02-11:
I agree, as a new user it was confusing.

If you don't know, you can clone all the wiki pages at once. There is a button at the bottom of the menu:
[image]

NoahAndrews, 2017-02-11:
I do agree that we need to require documentation before accepting new features.

jackhumbert, 2017-02-11:
We discussed this in #759 - we're gonna stick with the wiki for now, but the organisation of it still needs done. Concerning the changes to the wiki with new features, the suggested `[PROPOSED CHANGE]` would work well. In PRs, you can link compared revisions [like this](https://github.com/qmk/qmk_firmware/wiki/Home/_compare/eadb1afbd0dd13a38edad8948bf2d71766bc5760...c3ebd9e61badc47a69143e9448fbfecce1027ede), which would serve well for approving the documentation (something that should be done before submitting).

priyadi, 2017-02-14:
What kind of organization do you have in mind? Maybe we can help in some way.

I find the current one page format overwhelming, especially in wiki format where in most cases we use online editor to make changes. So I'm proposing we split each feature in its own page. If a single big page is desirable, maybe we can automatically generate it.

jackhumbert, 2017-02-14:
Yeah, separate pages would help a lot - I like the idea of covering a topic briefly in Home (insomuch that it wouldn't need to be changed when modifications to the feature are made/added), and linking to a page that covers it more thoroughly. There's a lot of the tmk pages left over that could be cleaned-up/deleted as well.

jackhumbert, 2017-03-29:
I'm gonna close this for now, since the Wiki will always be a work in progress :) thanks for your thoughts folks!

closed 2017-03-29 by jackhumbert