qmk/qmk_firmware #1096: How to use bluetooth on HHKB?

IDN1945, 2017-02-13:
I just flash QMK to my HHKB, the problem is when I try to use the keyboard via bluetooth it doesn't register any keys (I connect it to my android phone) even though it's paired and connected. Before using QMK, I use TMK and the bluetooth works just fine. Any idea to solve this? I use Hasu BT Alt Controller Thanks

uberjay, 2017-05-09:
I'm seeing the same thing here. Seems as though the rn42 (the BT radio) specific support only exists in tmk. (see https://github.com/tmk/tmk_keyboard/tree/master/keyboard/hhkb/rn42)

blakedietz, 2017-05-21:
Well uh, that sucks. I didn't realize that. What sort of work would need to be done in order to get qmk support for the rn42?

blakedietz, 2017-05-21:
Just for reference, I too can confirm that I'm having the same problem. I'm able to get the keyboard to pair, keys just aren't registered. This is for macOS 10.12.5 (16F73). I would be willing to put some work in to get this fixed, I would just need a few pointers to make it happen.

uberjay, 2017-06-01:
Well, fwiw, I managed to cram the rn42 code into the qmk hhkb source tree and get everything linked up correctly. At least, it's working well enough that I'm posting this via BT using my freshly built firmware! :)

I think I'll have a little time over the next couple days to clean it up and submit a proper PR. Stay tuned.

blakedietz, 2017-06-01:
Let me know if theres anything I can do to help. I can demo it on my board for a few days if you'd like.

blakedietz, 2017-06-10:
@uberjay any updates on this one? Do you have a private repo with these changes?

uberjay, 2017-06-11:
Ok, I cleaned up the changes a little, and pushed my bits to a branch. If you check out the `hhkb-rn42-from-tmk` branch from https://github.com/uberjay/qmk_firmware, you should be able to give it a try. I'm super unfamiliar with both qmk and tmk, so don't actually know what the right way to port these changes over is. See keyboards/hhkb/rules.mk for how to enable the rn42 support:

https://github.com/uberjay/qmk_firmware/blob/hhkb-rn42-from-tmk/keyboards/hhkb/rules.mk#L58

fredizzimo, 2017-06-11:
Hi @uberjay,

I quickly checked the integration, and I didn't find any major problems, so I suggest that you create a pull request.

The question is do we want to make this hhkb specific? I think not, and in that case the code should be moved to `tmk_core/protocol`. But that can be done later in another pull request. What do you think, @jackhumbert?

uberjay, 2017-06-11:
Hey @fredizzimo -- thanks for taking a look at it! That's one of the questions I had, too, but I didn't know enough about the hardware (or the software, really :) to make that call. What I can say is the rn42 support in https://github.com/tmk/tmk_keyboard is in the keyboard/hhkb directory. Perhaps @tmk can shed some light on this?

Another thing that gave me pause was finding a handful of references to *other* rn42 support in the tmk_core bits, introduced in commit 71da013995ec72293ab1f3e2518a99d64ec486fb just a couple months ago. But the changes aren't very extensive, so it's not clear to me if they should have worked. I tried building a stock qmk with `BLUETOOTH_ENABLE=yes` and `BLUETOOTH=RN42`, but it didn't do the trick for my BT HHKB module.

uberjay, 2017-06-11:
Ohhh, interesting. After poking around and reading about the HC-05 module referenced in that commit, I see that this is actually for supporting a different module that's been re-flashed with firmware previously extracted from an rn-42 module... so given all that, I have no idea if the existing code changes in, e.g., https://github.com/qmk/qmk_firmware/commit/71da013995ec72293ab1f3e2518a99d64ec486fb#diff-81c48611e68de700597fb5c837914039 *should* work with the actual rn42 module in hasu's hardware, or... yeah. I'm totally in over my head here. *grin*

blakedietz, 2017-06-11:
I can confirm that @uberjay's aforementioned branch works with my wireless hasu controller. Nice job!

jackhumbert, 2017-06-11:
Nice! Yeah, I'd love for this to be a core feature that others could include in their projects, but just having it for the HHKB is a good start :)

blakedietz, 2017-06-11:
Something to note for for awareness, I'm not complaining here :D, but the size of the build is quite large including rn42. This is to no surprise. With my layout I'm getting a build size of about 96% of the aloted memory for the controller. I ended up having to remove tapdance and unicode from my keymaps in order to get this to work, which is totally fine (I was planning on removing td anyway).

On another note:

@uberjay can I buy you a beer? DM me on twitter @dietztweetz so I can send you some beer money.

hugomd, 2017-09-12:
Sorry to bump this, but are there any plans for this to be merged in?

uberjay, 2017-09-26:
I'm going to try to clean this up and get a PR together over the weekend!

patrickhener, 2017-12-07:
Well I seem to not getting this thing to work. I don't know what I did wrong. i cloned your branch and then I copied my keymap in. I then compiled like: make patrick. It did the hex file. But after flashing it there is no change. I can connect but it will not register any keys. Did I do something wrong?

Enochen, 2017-12-22:
Any updates on this feature?

drashna, 2018-03-25:
Still looks like this works.  If somebody can still confirm that this works on an updated version,  and make a PR, that would be awesome

Enochen, 2018-04-06:
Totally forgot about this haha

It still works, just made a PR
https://github.com/qmk/qmk_firmware/pull/2693

BenKesselring, 2018-05-10:
Question: For the hhkb, is the RN42 the controller, or is it just a BT module connected to the controller? If the latter, would this make it possible to set up BT on _any_ keyboard using an RN42 module?

shelaf, 2018-05-10:
Hi @BenKesselring
The latter. cf. https://github.com/tmk/HHKB_controller/blob/master/schematic/HHKB_controller_revG.pdf
You can use RN42 module to set up BT, but it needs some knowledge about hardware hacking.

BenKesselring, 2018-05-12:
Thanks for the info! That's great news.