qmk/qmk_firmware #1106: Question: Is there any way to make LSFT_T(KC_FN1) work?

phongnh, 2017-02-16:
I wanted to map left shift as dual role key. Tapping it will switch layer 1, holding it is modifier key. I tried to use LSFT_T(KC_FN1) and defined fn_actions as follows, but it did not work

const uint16_t PROGMEM fn_actions[] = {
    [1] = ACTION_LAYER_TOGGLE(_SYMBOL),
};

Is there any way to make it work?

Thanks.

h-youhei, 2017-03-19:
I think it can be done by something like this.
But I don't know if this works.
```c
bool is_hold(keyrecord_t *record) {
  return (record->tap.count <= 0 || record->tap.interrupted);
}

const macro_t *tap_fn_hold_mod(keyrecord_t *record) {
  if (record->event.pressed) {
    if (is_hold(record)) {
      register_mods(MOD_LSFT);
    }
  }
  else {
    if (is_hold(record)) {
      unregister_mods(MOD_LSFT);
    else {
      // I don't know using this function is right.
      layer_invert(1);
    }
  }
};

void action_function(keyrecord_t *record, uint8_t id, uint8_t opt) {
  switch(id) {
    case _SYMBOL :
      tap_fn_hold_shift(record);
      break;
    default:
      break;
};

const uint16_t PROGMEM fn<actions[] = {
    [1] = ACTION_FUNCTION_TAP(_SYMBOL)
const uint16_t 

```

Reference

doc/keymap.md
- 1.5 Fn key
- 2.1 Key Action
- 2.4 Function action

tmk_core/common
- action_layer.h
- action_code.h

tmk_core/common/action.c
- case ACT_LMODS_TAP
  - default
- case ACT_LAYER_TAP
  - case OP_TAP_TOGGLE

phongnh, 2017-04-01:
Thanks @h-youhei. I will try it.

closed 2017-04-01 by phongnh