qmk/qmk_firmware #1109: Flashing firmware leaves keyboard unresponsive

djl, 2017-02-17:
This is a weird one.

I'm running into an issue where flashing the firmware for my Ergodox EZ *sometimes* results in an usable keyboard. Here's what happens:

    $ make ergodox-ez-djl-teensy
    ... SNIP ...
    Read "./.build/ergodox_ez_djl.hex": 22576 bytes, 70.0% usage
    Found HalfKay Bootloader
    Programming.................................................................................................................................................................................
    Booting
    $

Then the keyboard fails to boot at all. No flashing lights, no keys recognized. When checking Console.app, I see these messages:

    # immediately after "booting"
    2/17/17 11:38:26.000 kernel[0]: 042900.346705 IOUSBHostHIDDevice@fd330000,3: IOUSBHostHIDDevice::interruptRetry: resetting device due to IO failures

    # Keeps repeating until I manage to re-flash the firmware or unplug the keyboard
    2/17/17 11:38:36.000 kernel[0]: 042910.402082 AppleUSB20HubPort@fd330000: AppleUSBHostPort::disconnect: persistent enumeration failures
    2/17/17 11:38:38.000 kernel[0]: 042911.603210 AppleUSB20HubPort@fd330000: AppleUSBHostPort::disconnect: persistent enumeration failures
    2/17/17 11:38:38.000 kernel[0]: 042912.514334 AppleUSB20HubPort@fd330000: AppleUSBHostPort::disconnect: persistent enumeration failures
    2/17/17 11:38:42.000 kernel[0]: 042916.093613 AppleUSB20HubPort@fd330000: AppleUSBHostPort::disconnect: persistent enumeration failures
    2/17/17 11:38:43.000 kernel[0]: 042916.892220 AppleUSB20HubPort@fd330000: AppleUSBHostPort::disconnect: persistent enumeration failures
    2/17/17 11:38:43.000 kernel[0]: 042917.284211 AppleUSB20HubPort@fd330000: AppleUSBHostPort::disconnect: persistent enumeration failures
    2/17/17 11:38:44.000 kernel[0]: 042918.485709 AppleUSB20HubPort@fd330000: AppleUSBHostPort::disconnect: persistent enumeration failures

Repeatedly flashing the firmware eventually results in a usable keyboard. Usually five or six attempts does the trick.

This started happening a couple of weeks ago. I've attempted to `git bisect` to find the commit that may have caused this but I've yet to find anything.

Things I've ruled out:

1. USB connection. It happens on multiple USB ports, with and without a hub
2. My keymap. The issue persists with the default keymap

Things I cannot rule out yet:

1. Whether this is specific to macOS.
2. If this is a hardware issue with the keyboard itself

Beyond this I'm not sure where to start debugging. Any suggestions?

alazarsc, 2017-02-17:
What size is your file? Too large a file won't flash the board (though that should be repeatable 100% of the time, so not sure what's up with that)

Cribbit, 2017-02-17:
What size is your file? Too large a file won't flash the board (though that should be repeatable 100% of the time, so not sure what's up with that)

djl, 2017-02-18:
Both of the keymaps I tested are small enough to flash:

    Read "./.build/ergodox_ez_default.hex": 23066 bytes, 71.5% usage
    Read "./.build/ergodox_ez_djl.hex": 22576 bytes, 70.0% usage

unxmaal, 2017-02-21:
This sounds very similar to my experience flashing a ps2avrGB board.

I'd given up, assuming something was wrong with the PCB.

jmdaly, 2017-03-01:
I've got a RoadKit running QMK connected to a 2015 MacBook Pro and, although I haven't had any issues flashing, the board will stop working intermittently. I see similar message to you in the kernel logs:

`211079.738075 IOUSBHostHIDDevice@14140000,1: IOUSBHostHIDDevice::interruptRetry: resetting device due to IO failures`

The only way to get the board working again is to unplug and plug back in. I did some Googling, and found similar reports with USB issues in OS X [here](http://blog.michael.kuron-germany.de/2017/01/) and [here](https://www.reddit.com/r/applehelp/comments/3tj71w/usb_mechanical_keyboard_stops_working_after_el/?st=izrfm4l8&sh=7e12e9c1). So it looks like it may be a USB driver issue on the OS X side?  I'm going to plug the board into a Linux box and run it there for a while to see if there are any problems.

Hope this helps a bit. I'd be interested to know if you learn any more about the issue.

jaydpage, 2017-03-04:
@djl I have a similar problem also with an Ergodox EZ. I have not tried flashing it 6 times, however when I flash it with a layout generated from the massdrop configurator it works fine. I really want my QMK layout on my board so I am going to try repeatedly to see if I can also get it working as you did. Will let you know how it goes

unxmaal, 2017-03-04:
After more experimentation, this ultimately ended up working, without any modifications, on a fresh PCB. 

I suspect the original PCB arrived damaged. 

Closing.

djl, 2017-03-04:
This week I managed to test a few different setups. Here's the results:

- macOS El Capitan machine + Ergodox EZ == BAD
- macOS El Capitan machine + Ergodox original == BAD
- macOS Sierra machine + Ergodox EZ == BAD
- macOS Sierra machine + Ergodox original == BAD
- Linux + Ergodox EZ == GOOD
- Linux + Ergodox original == GOOD

I had zero issues flashing the firmware with Linux so I'm inclined to agree with @jmdaly. This seems like a USB driver issue with macOS.

macOS won't be my main OS much longer so I can't dedicate a ton of time to debugging this but I'd still like to see it fixed (or even just find out what the underlying issue with the USB drivers is).

jmdaly, 2017-03-06:
@djl Thanks for sharing the test results! As an additional data point, I had my board plugged into a Linux box all weekend without any issues. If I learn any more about the root cause of this issue on OS X, I'll definitely post here.

skullydazed, 2017-06-24:
Is this still a problem for people? I've been flashing boards on both sierra and el capitan for a while and have never run into it.

cquiroz, 2017-06-24:
I have been flashing an atreus62 and an ergodox infinity on sierra many times in the last 2 months without seeing this

abraithwaite, 2017-10-23:
I may have just run into this issue.

abraithwaite, 2017-10-24:
Update: No dice on linux either.

https://gist.github.com/anonymous/e0f25a919a704700cdc613c23a0e485d

canatella, 2017-11-29:
Same problem here, working on Linux but not on MacOS Sierra

drashna, 2018-03-25:
Is this still an issue on the newer version of MacOS?

ryan1st, 2018-07-08:
Still a problem here.
On MacOS(Sierra v10.12.6) with Teensy, I have to reboot my mac everytime I flash fw of Ergodox EZ, as it becomes unresponsive.

After successful flashing, the keyboard reboots without problem and I can see the booting lamps. But then the keyboard becomes unreponsive. In the keyboard settings of Mac OS, it cannot detect any keyboard. 

I tried changing usb port but same thing happens. Booting lightings and then unresponsive. 

The wierd thing is that it happens most of the time and I can tell when it will happen by observing my mouse pointer!!

I'm using Apple's Magic Trackpad (via bluetooth) and the pointer movement becomes strange(or laggy. it move and stop and move and stop...) as if there's problem communicating via bluetooth. 

To summarize, after rebooting keyboard(after flashing) if my mouse pointer becomes strange, then the keyboard becomes unresponsive 100% of the times.

Then, I reboot my mac and the keyboard comes back. 
This is quite annoying :(

drashna, 2019-03-29:
Was this still an issue?

Erovia, 2020-03-12:
Closing this issue as it hasn't seen much activity recently.
If you still have related questions, feel free to reopen this case.

closed 2020-03-12 by Erovia