qmk/qmk_firmware #1111: Emulating mouse scroll with a keypress

Cribbit, 2017-02-17:
Page up/page down exists, but does not always do what I want it to in a given scenario. Often, I want the scrolling functionality of a mousewheel, which means moving my hand off my keyboard to my mouse.

Is this functionality already in the firmware? I can't seem to find it if it is, but there's a lot of stuff in QMK that I haven't gotten to use yet.

If it's not already there, has anyone looked into adding it? I would like to work to add this feature and am wondering if anyone already has research into accomplishing it.

NoahAndrews, 2017-02-17:
It's there. [Here's a list of all the mouse-related keycodes in the wiki](https://github.com/qmk/qmk_firmware/wiki/Keycodes#mousekey)

closed 2017-02-17 by Cribbit

jackhumbert, 2018-03-18:
[Up-to-date link in the docs](https://docs.qmk.fm/feature_mouse_keys.html) (if this isn't up-to-date please comment here so I can edit!)

0x4C4A, 2018-08-03:
Link's stale, Jack - this one works now https://docs.qmk.fm/#/feature_mouse_keys?id=mousekeys