qmk/qmk_firmware #1114: single key for layer and modifier possible?

szymonkaliski, 2017-02-21:
I'm using HYPER key with Hammerspoon for custom functionality (like hyper + enter opens iterm, etc), I'd love to also use this key to change 1-10 into F1-F10, I could do that easily if that key would toggle a layer, but is it possible to have both?

So holding hyper and clicking 1-10 sends F1-F10, but holding hyper and clicking any other key sends hyper + that key?

eyJhb, 2017-03-22:
Make a new layer that has everything as transparent except 1-10 that'll be f1-f10?

rask, 2017-03-22:
But he wants the Hyper _modifier_ to act as a modifier and still have a `MO()` layer on it, if I understood correctly.

szymonkaliski, 2017-03-23:
Yeah, I want HYPER+a to send ctrl+cmd+alt+whateveritwas+a and HYPER+1 send just F1...

h-youhei, 2017-06-01:
I think ACTION_LAYER_MODS is for it.
I'm not sure because I haven't used the feature.

This might help you. #1352

drashna, 2018-03-25:
You may be able to do this now. #2460 added layer+mod support. 
https://docs.qmk.fm/feature_advanced_keycodes.html#switching-and-toggling-layers

Specifically, you want `LM(layer, mod)`

sandric, 2018-04-21:
@drashna no, its not, I just tried it, as I understood its just alias for ACTION_LAYER_MODS as @h-youhei pointed out, so its masking whole layer with modifier whereas question was how to make it work for transparents in layer and no mod for others. I just installed qmk because of this and turned out there's nothing there, unfortunately. @szymonkaliski have you figured out how to make it work on your example.

jackhumbert, 2020-03-13:
I think we can close this - implementing this with `process_record_user` seems pretty straight-forward. If anyone has code they can share as an example, that would be great.

closed 2020-03-13 by jackhumbert