qmk/qmk_firmware #1128: How to write a keymap for the Alphabet-layout Infinity 60%?

stites, 2017-03-02:
I noticed that they keymaps in `keyboards/infinity60` (header files and all) are based on a different layout than the Alphabet layout. Looking at Input Club's configurator I see three variants: Hacker, Alphabet, and Standard.

What would be the proper way to configure one of the non-standard layouts? Should I just pretend that some keys do not exist?

eulerdisk, 2017-03-04:
Yes, I just did for myself. I have the exact same keyboard and layout.

stites, 2017-03-08:
I just tried flashing the default layout onto my board with no success. @eulerdisk just as a sanity check, how did you flash? I am running the following:

```
cd keyboards/infinity60
make default-dfu-util # this errors because dfu-util is also finding the a runtime
dfu-util -d 11c1:b007 -D ../../.build/infinity60_default.bin -R
```

Does this look correct? Half of the keys are broken and the others are all out of order.

jpetermans, 2017-03-21:
@stites - I tested the default layout on my infinity and it works fine.  Did you define INFINITY_LED or INFINITY_PROTOTYPE in your config.h?  The prototype and production boards use a different matrix and pinout than the newer LED version of the pcb.

eulerdisk, 2017-03-21:
@stites I had the same problem. You have to undefine INFINITY_PROTOTYPE.

stites, 2017-04-11:
ahha! everything works! Thanks so much @jpetermans, @eulerdisk!

closed 2017-04-11 by stites