qmk/qmk_firmware #1135: Default rules.mk has wrong value for at90usb1286 MCU?

sdziscool, 2017-03-04:
For me it created:
```
# MCU name
#MCU = at90usb1287
MCU = atmega32u4
```
And as a newbie in the setup process I added A # to the 3rd line removed the # in front of `#MCU = at90usb1287` and after some hours of keymapping I tried to compile and it gave me weird errors in the quantum_keycodes.h file. After spending 2 hours trial and error I found that the teensy 2.0++ used 'MCU = at90usb1286' and then it worked perfectly again, but there is still something wrong with `MCU = at90usb1287`

Would be really nice if there was an easier way to retrace such an error, and something should probably be fixed somewhere.

jackhumbert, 2017-03-29:
Yeah, while the 1287 is an MCU, I'm not sure why it's being used there instead of the 1286, which is much more popular from what I've seen. I think this error (if it is one) came from TMK, but I'd be happy to switch everything over the 1286!

sdziscool, 2017-03-29:
Probably for the best, seeing as 1287 doesn't seem to be (fully) supported. It gave a **lot** of errors when compiling for me, but turning the 7 into a 6 made everything work perfectly so I suppose it was never meant to be 1287, otherwise the 1286 should have also been an option in the comments.

commit 582a6ac7 (2017-06-24): Fix #1135 by changing the default to at90usb1286
closed 2017-06-24 by skullydazed