qmk/qmk_firmware #1141: Switching from LEGACY_KEYMAP to hash causes spacebar to freak out

ajacksified, 2017-03-07:
I was working on programming my s-60x; besides a few changes to how the arrow keys get mapped, I also wanted my caps lock to map to ctrl on hold and esc on tap. However, whenever I switch from LEGACY_KEYMAP to a hash, my spacebar sends F19 (0x6e), and everything to the right doesn't register any keypresses at all.

My keymap: https://github.com/ajacksified/qmk_firmware/commit/6e855632e80be6728fbf596663b6b2abcb2dab5e

Is this an issue anyone has seen before?

fredizzimo, 2017-06-24:
I'm don't know what you mean by "a hash". But did you try the normal `KEYMAP` instead of `LEGACY_KEYMAP`? By doing that you need to add KC in front of most symbols, so `A` becomes `KC_A`.

drashna, 2018-03-25:
Is this still an issue?

closed 2018-10-21 by drashna