qmk/qmk_firmware #1153: mis behaver with GUI_T(KC_LPRN) or all the special character with "shift" key

dunkarooftop, 2017-03-10:
Want to setup LGUI when tap show "(" and act as "LGUI" when held

GUI_T(KC_LPRN) send "9" to the os when tap and "cmd + shift" when held.

tested on OSX

Avispa, 2017-04-09:
Same for CTL_T(DE_RCBR) and ALT_T(DE_RBRC). Former sends "0" and latter "9" which are in fact CTL_T(RALT(KC_0)) and ALT_T(RALT(KC_9)) . Have it on a momentary layer but referring to the post before it seems it doesn't matter.

Sadly it seems I need them on the momentary layer to get out of the momentary while holding Ctrl and/or Alt from the top layer.

dunkarooftop, 2017-04-10:
I end up use my function from TMK to make it happen. Here is the code

```
void action_function(keyrecord_t *record, uint8_t id, uint8_t opt)
{
    if (record->event.pressed) dprint("P"); else dprint("R");
    dprintf("%d", record->tap.count);
    if (record->tap.interrupted) dprint("i");
    dprint("\n");

    switch (id) {

    case LGUI_LPAREN:
        if (record->event.pressed) {
            if (record->tap.count > 0 && !record->tap.interrupted) {
               if (record->tap.interrupted) {
                   dprint("tap interrupted\n");
                   register_mods(MOD_BIT(KC_LGUI));
               }
            } else {
                register_mods(MOD_BIT(KC_LGUI));
            }
        } else {
            if (record->tap.count > 0 && !(record->tap.interrupted)) {
                add_weak_mods(MOD_BIT(KC_LSHIFT));
                send_keyboard_report();
                register_code(KC_9);
                unregister_code(KC_9);
                del_weak_mods(MOD_BIT(KC_LSHIFT));
                send_keyboard_report();
                record->tap.count = 0;  // ad hoc: cancel tap
            } else {
                unregister_mods(MOD_BIT(KC_LGUI));
            }
        }
        break;
    }
}
```

closed 2017-06-24 by skullydazed