qmk/qmk_firmware #1156: Can I use this¿

XenGi, 2017-03-11:


armatus, 2017-03-12:
surely, but for WiFi toggle and brightness you must register shortcut in your OS
(there is no keycodes for such things)
if KB hand-wired, you may need 19+6 pins (Teensy 2.0 or Teensy++ 2.0) as micro-controller
it's still possible use cheap ProMicro with IO expander

XenGi, 2017-03-12:
I use a Teensy 3.0 so this should work.
The problem I'm facing is that my layout is not a supported one, so I will need to do more then just define my matrix and that's it.
I can't figure out if this is even possible or if I will most likely need to completely rewrite most of qmk to work with my keyboard.

I'm mostly concerned about the way my NumLock works. If it's off I have a tenkeyless keyboard but if it's on the arrow keys etc turn into a numpad. This needs to be handled in the firmware and I haven't seen a keyboard doing that, except the Coolermaster one where I got the idea.

armatus, 2017-03-13:
it's no need to rewrite QMK. look at any keyboard in /keyboards directory of sources (./planck for example) 
define your pins in config.h
`
#define MATRIX_ROWS 4
#define MATRIX_COLS 12

#define MATRIX_ROW_PINS { D0, D5, B5, B6 }
#define MATRIX_COL_PINS { F1, F0, B0, C7, F4, F5, F6, F7, D4, D6, B4, D7 }
`
look how to map keymap to matrix aliases in planck.h

only one left - write your own keymap.c

ShetiPhian, 2017-03-13:
I've only been playing with the firmware for a couple of days, so there might be a better way.

You'd set up three layers; (Layer0 - Default, Layer1 - Numpad, Layer2 - Function)
Your NumLock key would be a toggled layer switch, toggling Layer1
Your FN key would be a momentary layer switch, activating Layer2 while held

Just a heads up:
I'm also using the Teensy 3, while the code compiles and runs providing a usable keyboard, I've ran into a few issues.
- Cant use WS2812 RGB underlight, enabling the option stops the firmware from compiling
- Keyboard always shows up a a generic HID device with unknown manufacture regardless what you set in config.h

armatus, 2017-03-13:
i'm don't have a such issue, but i'm using teensy 2.0
create a new issue about that with detailed logs

closed 2017-06-24 by skullydazed