qmk/qmk_firmware #1160: Momentarily set default layer

27percent, 2017-03-13:
Hi there,

Looking through some of the planck keymaps, I noticed rather than turning layers on or off, you're setting the default layer. I think this is a great idea, and it makes way more sense to people new to mechanical keyboards to say you're switching to a layer vs teaching them about the layer stack. 

I was wondering if you've had any ideas on implementing a momentary alternative to the switching you're doing right now - so switch the default layer while the key is held, then switch it back to the previous default layer when the key is released.

I've been hacking around with it for a little while but I've not made much progress, your thoughts on a solution would be great to hear if you have them - happy to implement and submit a pull request if you can point me in the right direction :)

27percent, 2017-04-02:
@eyJhb: It is indeed, but only for toggling layers, I'm looking to set the default layer instead so the user can move both up and down the layer stack. Also, TG/LT wouldn't allow the user to attach custom actions to layer switching (like changing the backlight brightness for example) :)

Avispa, 2017-06-13:
Not sure, but I think I have a similar problem. So far I used `DF(...)` to switch to the desired layer (to change the keyboard layout). Related to my current investigation that screws `visiualizer.c` because it never notice a change. TG or LG is not what I am looking for -- I don't wanna have a second key on top nor a toggle.

Side note, even though I have the idea I read it already somewhere, with `TG(...)` only the LCD on the keyboard half where the `TG(...)` key is displays the layer change. (ErgoDox Infintiy)

fredizzimo, 2017-06-13:
@Avispa, changing the default layer is supposed to work correctly with the visualizer. 

So there's either a bug in the visualizer or the your implementation. I can investigate that, if you have a link to your keymap.

drashna, 2018-10-21:
There may need bo something called in `default_layer_state_set_user`.

jackhumbert, 2020-03-13:
I think this could be accomplished with `process_record_user` and a custom keycode that sets the default layer on press/release. I think we can close this for now, but if anyone has code to share that implements something similar, that would be great.

closed 2020-03-13 by jackhumbert