qmk/qmk_firmware #1164: Send extra info with macro

PureSpider, 2017-03-17:
Is it possible to attach some piece of meta information to an `M()` call in my keymap, other than the macro id?

jackhumbert, 2017-03-17:
I've been advising people to try out [`process_record_user(uint16_t keycode, keyrecord_t *record)`](https://github.com/qmk/qmk_firmware/wiki#bool-process_record_uint16_t-keycode-keyrecord_t-record) and encoding their own information in their keycodes :)

PureSpider, 2017-03-17:
How would I go on using this to have something like `X(name, keyword)` where name is just the name and keyword is said extra meta information?

jackhumbert, 2017-03-17:
It'd probably be easiest to set up an enum for `name` and `keyword`, and use those to pass information into the function. How many keycodes are you looking to use in total?

`SAFE_RANGE` is the first keycode that's not used by QMK, so you can use that as your basis for defining them, but if you're going to encode info into them, you'll probably want to jump up to the closest block so you can do stuff like:

    #define MY_THING(name, keyword) ( SAFE_BLOCK | (name) | (keyword << 2) )

This is how a lot of the built-in functions are defined.

PureSpider, 2017-03-17:
That's some very useful insight! I'll see if I can get this to work, thank you!
What's the maximal length of name and keyword combined I could encode into that?

jackhumbert, 2017-03-17:
I'm not sure - you could try outputting `SAFE_RANGE` to see where it starts :) there's 0xFFFF total right now.

PureSpider, 2017-03-17:
To make all this a little less abstract: I'm trying out my own idea of unicode input and I want to route it over one function that essentially is something like X(unicode) and then uses that unicode information to send it to the host. Since I want to go beyond the 0x7FFF limitation, I'd need some space to encode the characters.

jackhumbert, 2017-03-17:
There's quite a few unicode implementations - I'm not familiar with all of them/their uses, but is there anything there that could be used?

I've found that the easiest way to implement unicode on a personal level is to just define custom keycodes for each character with a listener in `process_record`, and place those in your keymap.

PureSpider, 2017-03-17:
I checked and I am not happy with either implementation. I have my keyboard talking to the host over HID already either way, so I'll try and use that channel to send unicode to the host and have a small programm running there do the actual unicode input.
For that to work, however, I'd need some way to call my "send over HID" function from the keymap/keypress with some meta information on what to send attached.

drashna, 2018-03-25:
That block code stuff could use documentation (by somebody that understands it better).

As for the unicode, I'm not sure how much has changed since ... but:
https://docs.qmk.fm/feature_unicode.html

drashna, 2018-10-21:
Sounds like this is safe to close now.

If not, please reopen.

closed 2018-10-21 by drashna