qmk/qmk_firmware #1165: Unicode Input with WinCompose fails because of it is case-sensitive

nishio, 2017-03-19:
WinCompoese under default settings doesn't accept unicode hex inputs because it suppose U and A~F are all uppercase.
However hex_to_keycode in quantum.c (called from process_unicode.c) generate lowercase keycodes.

https://github.com/qmk/qmk_firmware/blob/b794719952edafa174a1eea031c3433be820ad29/quantum/quantum.c#L1015

To Configure WinCompose to case insensitive works as a walk-around.

drashna, 2018-10-21:
This function is defined weakly, so you could replace this in your keymap to handle it "correctly".

If that does work, then it may be worth opening a PR that changes the default functionality.

drashna, 2019-03-29:
Unicode was significantly overhauled a bit ago.  This issue may no longer be an issue.

nishio, 2019-03-29:
Thank you. Should I close the issue?

drashna, 2019-03-29:
Yeah. I can do so too.  But I would recommend updating your repo and WinCompose, and testing the changed code, just to make sure it fixes the issue. 

If it does, close the issue, if not, let us know.

closed 2020-03-13 by jackhumbert