qmk/qmk_firmware #1166: Right-hand modifiers are not available for `MT()`

paldepind, 2017-03-19:
The wiki states:

> These are the values you can use for the `mod` in `MT()` and `OSM()` (right-hand modifiers are not available for `MT()`):
>  * MOD_LCTL
>    * MOD_LSFT
>    * MOD_LALT
>    * MOD_LGUI
>    * MOD_HYPR
>    * MOD_MEH

However, I would like to create a key that functions as enter when pressed alone and right alt when held.

Can the restriction be lifted so that `MT` also works with right-hand modifiers?

priyadi, 2017-03-19:
It should work. Try `MT(MOD_RALT, KC_ENT)`.

The documentation hasn't been updated yet.

paldepind, 2017-03-19:
@priyadi Thank. You that is wonderful. It works splendidly :smile:

It that the case for all the right-hand modifiers? Should I update the wiki?

priyadi, 2017-03-19:
Please do if you can.

It should work for all modifier (CTL, ALT, GUI), but IIRC limited only to one side. You can't mix left and right side modifiers. If you do, the right side takes effect.

paldepind, 2017-03-19:
I have updated the wiki [here](https://github.com/qmk/qmk_firmware/wiki/Key-Functions).

I'm happy to have enter on my right alt :+1:

closed 2017-03-19 by paldepind