qmk/qmk_firmware #1179: Backlight support for Infinity60

jpetermans, 2017-03-28:
Is there any backlight support for the Infinity 60% (1.1 LED revision)? It uses the same controller as the whitefox and ergodox infinity (IS31FL3731C). It seems that individual switch level control is possible. The pinouts are available here: https://github.com/kiibohd/pcb/blob/master/IC60/IC60.pdf. But I don't see an obvious way in qmk of enabling the backlighting in any form, even a simple BL_TOGG or CapsLock. 

The IS31 code I've perused is way beyond my feeble coding skills, but I'd be happy just to enable backlighting or Caps/Num Lock leds. Is there anything that can be ported over from the ergodox folders? I can hack away at something but could use some direction and a pretty good head start with some template code.

fredizzimo, 2017-03-30:
I think once I get things working for the Infinity Ergodox, it should be quite easy to port to the Infinity60 as well as they are using the same chip. And yes that includes full control for each LED.

jpetermans, 2017-04-01:
@fredizzimo - Awesome. Let me know if I can assist with testing or whatever. Happy to help out where I can.

Does the whitefox tmk code have any portability? I've looked through it and understand the basics of it, but it seems to require some additions to tmk_core. The qmk [readme](https://github.com/qmk/qmk_firmware/blob/master/keyboards/whitefox/readme.md) for the whitefox suggests it's not worth porting over to qmk.

fredizzimo, 2017-04-01:
Yes, the same applies for the WhiteFox, it also uses the same LED controller, and runs ChibiOS, so it should be quite easy to port it to that as well.

I think the WhiteFox LED code could be ported quite directly from TMK, but it's very direct to the metal, and hard to use. I don't think it's that much of a hack, rather than making something simple that just works if you are able to work on that level. The hardware also provide some things, like built in animation support, audio synchronization and breathing, which my approach doesn't.

Except for the audio synchronization, the visualizer and in the future the effect system still allows you to write breathing and animiations, but in software instead. The processors on these keyboards are fast enough for that. My code exposes the LEDs as two [uGfx displays](https://wiki.ugfx.io/index.php/GDISP), one for each side. So in addition to setting single LEDs(pixels), you can for example also draw lines, or even display text.

jpetermans, 2017-04-03:
Thanks for your responses. You've encouraged me to take a shot at porting over the whitefox tmk code. I've made some progress and will report back if I manage to get any lights to turn on.

val3nt33n, 2017-05-08:
I was also looking for some basic LED support on Infinity 60% leading me to this thread.
Anyway, I wrote something quickly to implement some functionality that I need. It's not optimal but it works very well on Infinity 60%.

It supports backlight on/off increase/decrease, caps/num/scroll lock blinking regardless of the backlight setting, individual leds, and switching to a layer will light up only keys in that layer.

Here's the code:
https://github.com/val3nt33n/qmk_firmware/tree/infinity60/keyboards/infinity60

jpetermans, 2017-05-08:
@val3nt33n thanks for sharing your led code. I'm about 95% through my own revision. I'll definitely look through your led files. I've got a lot of the same functionality but yours looks very clean.

jpetermans, 2017-05-08:
Below is my code. I put more onus on the user keymap.c to figure out when and how to set leds, but I wanted to keep things flexible. This is my first go with c and chibios, so input is welcome.

[https://github.com/jpetermans/qmk_firmware/tree/user_led_configuration/keyboards/infinity60](https://github.com/jpetermans/qmk_firmware/tree/user_led_configuration/keyboards/infinity60)

readme is under my keymaps folder:
[https://github.com/jpetermans/qmk_firmware/tree/user_led_configuration/keyboards/infinity60/keymaps/jpetermans](https://github.com/jpetermans/qmk_firmware/tree/user_led_configuration/keyboards/infinity60/keymaps/jpetermans)

jpetermans, 2017-05-10:
@jackhumbert, I've written some backlight support for the infinity60 that I think is worth including in the infinity60 project folder. It's working fine on my board. I also have a custom keymap and readme that demonstrates and summarizes the options available; see my prior comment above.

Since not all infinity60's come with led functionality, I'm not sure if this would be best merged into the project folder with makefile comments or just kept as custom files in a keymap folder. Currently, my approach has led code in the project folder with a custom keymap folder that utilizes the new backlight code.

Happy to do whatever you think is most efficient to make qmk backlight support available to others for this keyboard.

shohart, 2017-06-30:
Guys, i am soory to interrupt with such a noobish questions, but if it is any way to implement such led functionality into GH60 satan build?

jpetermans, 2017-07-06:
@shohart I don't know what led controller the GH60 uses. The led code for the Infinity60 is very specific to this, whitefox and the ergodox pcbs. The effect system that is coming together may be more generally applicable, but I'm not very familiar with it yet.

I don't think anything under this issue will be helpful for you, and since led backlighting is now built into QMK, I'll close this issue.

closed 2017-07-06 by jpetermans