qmk/qmk_firmware #1181: Which is the Teensy Linux Page?

Rovanion, 2017-03-28:
On line 72 of `keyboards/ergodox/readme.md` it's mentioned that one should modify some udev rules as described on some Teensy Linux Page, but the page itself is not linked so the user is left in the dirt. Which is this page specifically?

Going through the git blame of the readme I see that @linsomniac introduced this line to the page, perhaps you can answer this question? Before `9e03517ae51ff30d332accae27469ae3b1b25e73` there are no mentions of udev in the readme.

Rovanion, 2017-03-28:
Related PR waiting for answer on the subject: #1031.

exiva, 2017-03-28:
https://www.pjrc.com/teensy/loader_linux.html has the example udev rule, you may just need to modify the vendor and pid's in the udev rule to match the teensy you have.

closed 2017-03-29 by Rovanion