qmk/qmk_firmware #1186: QMK logo and branding

priyadi, 2017-03-30:
Create a logo and branding for QMK. Can be useful for QMK web site, products using QMK, slides and presentations on QMK, etc.

fredizzimo, 2017-03-30:
I was going to ask the same at some point, as it would also be nice to display the QMK logo on the LCD screen at the startup of the Infinity Ergodox for example.

But that would need a monochrome low resolution version as the screen is 128x32 and supports only one and off pixel control.

skullydazed, 2017-03-30:
@jackhumbert has been working on some of that. We have a psi (Ѱ) we've used a few places, like gitter and the firmware flasher.

jackhumbert, 2017-03-30:
Yeap! I have the main one I like on https://github.com/qmk - there's also [one just for the flasher as well](https://github.com/qmk/qmk_firmware_flasher/blob/master/build/icon.iconset/icon_256x256%402x.png) :)

@fredizzimo how's this? 
[image]

priyadi, 2017-03-30:
@jackhumbert 

Do you have the logo in SVG format?

Any guidelines if I want to mention the word "QMK Firmware" next to the logo? Proper wording? Font? Color? Position? Because as it is, the logo alone is not easily identifiable with QMK, especially for outsiders.

jackhumbert, 2017-03-30:
Sure! How are these things usually organised in repos? `/style`? I can add a folder with the svg, some raster images examples, and guidelines.

priyadi, 2017-03-30:
@jackhumbert  Simply putting brand files in a directory should be enough for now

Some examples of good branding guidelines for your consideration:

* git: https://git-scm.com/downloads/logos
* ubuntu: https://design.ubuntu.com/brand/ubuntu-logo

fredizzimo, 2017-03-31:
@jackhumbert I will test the logo during the weekend. 

I won't be able to get white on black though, at the best it will be light gray on black. I can also test with other colors, as almost anything else than white is achievable with the backlight, do you have any preferred alternative color for white?

fredizzimo, 2017-03-31:
I just tested the logo on the LCD, it works great, thanks.  There's no need to change the backlight color or anything, a middle brightness is good.

Avispa, 2017-04-12:
Stupid (intentional) question but how did you came up with the Psi? Does't QMK means Quantum Mechanical Keyboard (Firmware)?

jackhumbert, 2017-04-12:
It describes the [wave function](https://en.wikipedia.org/wiki/Wave_function) in quantum mechanics :)

Avispa, 2017-04-12:
Okaaaaaay. Hope you got my point anyway. :)
Just in case, let me clarify. I am not sure how many of the (also potential) users have a degree in physics with a focus in quantum mechanics but I think by a logotyping/branding point of view to me the strategy of a logo which is easy to understand and easy to associate seems more appealing. When I need a degree in QM first, something goes wrong and I already have at least an education in physics. ;)

jackhumbert, 2017-04-12:
It's mostly trivia - the interpretation of the logo isn't that important for an open source project. QMK is just a play on words, and its logo is an extension of that :)

Avispa, 2017-04-12:
> logo isn't that important for an open source project.
How do you come up with this idea? The starter of the thread/issue proofs this statement already wrong. Even though there are multiple -- also scientific -- articles on which are working out that it is **at least** as important as for any other (financial) project. When you are interested I can search for a detailed and interesting article on this topic I was reading once.

> QMK is just a play on words, and its logo is an extension of that :)
Yes and no.
* Yes, a logo _should_ be an extension of the name.
* No, it is not "just" an extension; It is, from my point of view, already a step ahead and already an extension of the the logo of the name with a letter from Greek, thus no logo.

> It's mostly trivia
Maybe it is not? ;)

jackhumbert, 2017-04-12:
> the interpretation of the logo isn't that important for an open source project.

Is my full quote - the logo already existed, and the purpose of this issue is to now make more visible/usable for others. I'm not saying it's perfect, but [I think we're doing alright compared to similar projects](https://www.sporcle.com/games/mi0tch/open-source-software-and-language-logos-pt-ii). 

Your second part seems to be focusing on that the logo is an extension of the name, but my main point there was that it's an extension of the play on words, and "the joke."

This issue really isn't the place to be discussing any of this - let's keep future comments focusing on the implementation of the guidelines/ideas mentioned.

drashna, 2018-10-21:
@jackhumbert I think this can be closed?

closed 2020-03-13 by jackhumbert