qmk/qmk_firmware #1191: Split keyboard based on let's split firmware

xyxjj, 2017-04-01:
i've been working on my handwired split 65% keyboard, with a modified let's split firmware. most of the keyboard works already, just that the last column, column 8 doesn't work at all, and #define MATRIX_COLS 9 doesnt seem to have any effect on my keyboard. there doesnt seem to be any issues when generating the hex file so I'm not sure where to find and fix the problem

ahtn, 2017-04-03:
If you want to get >= 8 COLS, you need to modify some of the code ([as noted in the readme](https://github.com/qmk/qmk_firmware/blob/master/keyboards/lets_split/matrix.c#L51)).

Here's the relevant things you'd need to change (for serial):

* [Here](https://github.com/qmk/qmk_firmware/blob/da53f3a529e6e32473e67ae75021f28f9078d1e6/keyboards/lets_split/matrix.c#L56), the matrix is only using `uint8_t` -> `matrix_row_t`.
* Change [`serial_slave_buffer_length`](https://github.com/qmk/qmk_firmware/blob/da53f3a529e6e32473e67ae75021f28f9078d1e6/keyboards/lets_split/serial.h#L14) to `((MATRIX_COLS+7)/8 *MATRIX_ROWS/2)
* Change the master and slave code that transmits the matrix: [master](https://github.com/qmk/qmk_firmware/blob/da53f3a529e6e32473e67ae75021f28f9078d1e6/keyboards/lets_split/matrix.c#L195), [slave](https://github.com/qmk/qmk_firmware/blob/da53f3a529e6e32473e67ae75021f28f9078d1e6/keyboards/lets_split/matrix.c#L245). Just `memcpy` with the appropriate size.

There's probably some other thing's that I've missed, but that should get you started if you want to add this.

xyxjj, 2017-04-03:
thanks answering my question! unfortunately, im not very well versed in coding (i can code as well as a rock), so im not sure what you mean with regards to the last point

ahtn, 2017-04-03:
A non-coding solution would be to rewire your key matrix so that it only uses 8 columns. It'd be pretty easy, just wire the 9th column as a row instead.

benji877, 2017-06-08:
Could you elaborate more on your last point?

drashna, 2018-03-25:
@benji877 The software matrix doesn't have to match the hardware matrix.

You may want to read this:
https://docs.qmk.fm/understanding_qmk.html

But you would basically set the PCB so that it had less than 8 columns, and remap the wiring to match.  
It may/will make the PCB/wiring that much more complex, but it may be simpler than rewriting the code.

drashna, 2018-10-21:
Well, the orthodox is doing this now.   As is the quefrency from keeb.io. 

I think it may be safe to close this now. If not, please reopen.

closed 2018-10-21 by drashna