qmk/qmk_firmware #1192: [Preonic] GUI_T and ALT_T are not swapped when AG_SWAP is enabled

phongnh, 2017-04-01:
Currently, I use Preonic for both Windows and Mac. When I'm on Windows I press AG_SWAP to swap LALT and LGUI. However, only KC_LALT and KC_LGUI is swapped. GUI_T and ALT_T are still same.

FYI. Here is my keymap settings
```
/* Custom
 * ,-----------------------------------------------------------------------------------.
 * |   =  |   1  |   2  |   3  |   4  |   5  |   6  |   7  |   8  |   9  |   0  |  -   |
 * |------+------+------+------+------+------+------+------+------+------+------+------|
 * | Tab  |   Q  |   W  |   E  |   R  |   T  |   Y  |   U  |   I  |   O  |   P  | Bksp |
 * |------+------+------+------+------+-------------+------+------+------+------+------|
 * | Esc  |   A  |   S  |   D  |   F  |   G  |   H  |   J  |   K  |   L  |   ;  |  "   |
 * |------+------+------+------+------+------|------+------+------+------+------+------|
 * | Shift|   Z  |   X  |   C  |   V  |   B  |   N  |   M  |   ,  |   .  |   /  |Enter |
 * |------+------+------+------+------+------+------+------+------+------+------+------|
 * |RShift| Ctrl | Alt  | GUI  |Lower |    Space    |Raise | Left | Down |  Up  |Right |
 * `-----------------------------------------------------------------------------------'
 */
[_CUSTOM] = {
    {KC_EQL,         KC_1,        KC_2,    KC_3,    KC_4,  KC_5,   KC_6,   KC_7,  KC_8,    KC_9,    KC_0,                KC_MINS},
    {KC_TAB,         KC_Q,        KC_W,    KC_E,    KC_R,  KC_T,   KC_Y,   KC_U,  KC_I,    KC_O,    KC_P,                KC_BSPC},
    {ALT_T(KC_ESC),  KC_A,        KC_S,    KC_D,    KC_F,  KC_G,   KC_H,   KC_J,  KC_K,    KC_L,    LT(_MOUSE, KC_SCLN), GUI_T(KC_QUOT)},
    {KC_LSFT,        CTL_T(KC_Z), KC_X,    KC_C,    KC_V,  KC_B,   KC_N,   KC_M,  KC_COMM, KC_DOT,  CTL_T(KC_SLSH),      SFT_T(KC_ENT)},
    {RSFT_T(KC_GRV), KC_LCTL,     KC_LALT, KC_LGUI, LOWER, KC_SPC, KC_SPC, RAISE, KC_LEFT, KC_DOWN, KC_UP,               KC_RGHT}
},
```

jackhumbert, 2017-04-01:
Ah, yeah - the swapping isn't setup for those keys, but it could be! I'll see if I can't look into this next week. Thanks for bringing it up!

phongnh, 2017-04-01:
Thanks @jackhumbert

jackhumbert, 2017-04-03:
@phongnh would you mind testing #1202?

closed 2017-06-26 by jackhumbert