qmk/qmk_firmware #1193: LT(layer, kc) and Volume Keys Incompatible?

ShetiPhian, 2017-04-02:
It looks like the volume keys _KC_MUTE, KC_VOLU, KC_VOLD_ (and their alternative names) do not work with Layer Tap. (maybe all media keys?)

Originally I was using mute and it appeared as if the keyboard locked up.
Switching to volume up showed it was actually just stuck repeating the key until the keyboard was disconnected.

When I removed the layer tap and used the mute directly it work as it should, and when using another key like caps lock in the layer tap showed all was fine there (tapping toggled caps, holding activated the layer)

drashna, 2018-03-25:
Oh....  This is probably because you're on the wrong OS.  

If you're on macOS.... the keycodes may be different. 
KC__MUTE, KC__VOLUP, KC__VOLDOWN are the macOS codes:
https://docs.qmk.fm/keycodes.html

ShetiPhian, 2018-03-25:
Using windows,
I haven't worked on the board in a while, have to go by memory, the keymap, and original post.

On my Layer 0 Matrix, the mute button is defined as: F(2)
The media layer mute button is defined as: KC_TRNS

`const uint16_t fn_actions[] = {
  ACTION_LAYER_MOMENTARY(2),         // FN0
  ACTION_LAYER_TAP_KEY(3, KC_MENU),  // FN1
  ACTION_LAYER_TAP_KEY(4, KC_MUTE),  // FN2
  ACTION_LAYER_TOGGLE(1),            // FN3
};`

Trying to use layer tap would lockup the keyboard.
Changing KC_MUTE to KC_VOLU resulted in the same lockup, but with the volume maxing out and getting stuck there.
It turns out the keyboard get stuck holding down the volume keys.

Using KC_STOP for the layer tap allowed it to work without causing the keyboard to lockup.
Mapping the volume keys to the main layer worked correctly.

drashna, 2018-03-25:
I'm using LT(x,y), which an alias for this.  

However it does appear to work, when testing it.  And doesn't lock up the board
The tapping code may affect this, and tapping term definitely would. 
Maybe defining RETRO_TAPPING would help here?  

That and updating (merge or rebase) may help too, as the underlying issue may have been fixed.

drashna, 2018-10-21:
Ah, yes, only basic keycodes work, not the media keys. :(

closed 2018-10-21 by drashna