qmk/qmk_firmware #1200: is it possible to add pauses in dynamic macros?

sboesebeck, 2017-04-03:
following scenario on OSX: have a website open, want to print it (CMD-P), WAIT a second or so, then hit ENTER. Wait another second and hit CMD-W to close the window

If you want to do something like that with dynamic macros, which keycode do you need to include or how could that be possible?

Vifon, 2017-05-04:
Yes, it is possible. You need to create a new macro that will wait for the specified interval. See my keymap: https://github.com/qmk/qmk_firmware/blob/2704c163da590d3020f6d87717c400611de81110/keyboards/planck/keymaps/vifon/keymap.c#L167-L171

Alternatively you may tap some layer modifier a few times. Processing these "empty" keypresses will act as a pause. I prefer the first method though.

adzenith, 2017-05-11:
If you're using `MACRO()`, you can use `W()`to wait as described here:
https://github.com/qmk/qmk_firmware/wiki/Macros

jpetermans, 2017-05-11:
`_delay_ms` and `W()` work great for regular macros, but I don't think they can be used for dynamic macros that are typed  at runtime and temporarily saved.

Nephiel, 2017-05-11:
I just tried this. Defined a keycode in my keymap, switch-cased it in process_record_user, called _delay_ms() from there like Vifon did (though I return false after that), and it seems to work - both recording and playing back a dynamic macro.

Vifon, 2017-05-12:
It works because the keypresses replayed by the dynamic macros are handled just like the regular keypresses. There is only thing I can think of that is not recorded/replayed: the delays between the presses/releases.

And yes, `return false` makes sense here, though it won't hurt either way.

jpetermans, 2017-05-12:
Nice! I learned something new. Good stuff, and thanks for the correction. Seems this question has been answered.

closed 2017-06-24 by skullydazed