qmk/qmk_firmware #1205: kb / user hooks - TMK vs QMK

jpetermans, 2017-04-03:
I'm in the process of trying to port over matt3o's whitefox backlight code from TMK and converting for infinity60. In another issue I opened, #1179, fredizzimo commented that he is working on a software solution for the ergodox that could be ported over. In the meantime, I thought I'd take a stab at porting over the hardward approach from the whitefox tmk code. 

matt3o's led_controller code uses functions from tmk_core/common/hook.c which qmk does not have. I'm trying to translate to equivalent the kb / user level functions from qmk and want to make sure the timing of the calls is similar. My attempt is below.  Bolded calls are used in led_controller.c from tmk. Do these look correct? My primary concern is matching the hook_early_init timing.

- **Hook_early_init** - just after processor startup (**qmk = matrix_init**)
  - initialize I2C, is31_init, write led pages, clean up capslock register
- **Hook_usb_suspend_entry** - called once after suspend rec'd (**qmk = shutdown_user**)
  - enable sleep led
- **Hook_usb_suspend_loop** - called while in suspended state (**qmk = ??**)
  - check for usb activity and send wakeup
- **Hook_usb_wakeup** - called once after wakup rec'd (**qmk = startup_user**)
  - disable sleep led

(unused in whitefox led_controller code)
- Hook_late_init - last stage of init, just before keyboard loop (qmk = ??)
- Hook_keyboard_loop - called from keyboard loop (often) (qmk = matrix_scan)
- Hook_matrix_change - called matrix change event (keypress, often) (qmk = matrix_scan)
- Hook_default_layer_change - default layer change event (qmk = process_action_kb)
- Hook_layer_change - layer state change event (qmk = process_action_kb)
- Hook_keyboard_leds_change - indicator LED change event (qmk = led_set)
- Hook_bootmagic - called when checking bootmagic combo (qmk = ??)

jackhumbert, 2017-04-06:
I wouldn't mind setting up a framework for more hooks like this, or even renaming some of the existing ones. I wasn't aware of the hooks TMK added - this might have happened after we forked.

jpetermans, 2017-04-06:
The matrix_init_ and matrix_scan_ functions seem to cover most scenarios. matrix_init worked fine for the tmk backlight code I'm working on from whitefox. It was using hook_init_early. Although, additional hooks may be less processor intensive in some cases where you don't need to run something every keyboard loop, just on a few discrete events.

luizribeiro, 2017-04-09:
This would be nice to have. I've been using `matrix_scan_user` for changing backlight color when the current layer changes, but TMK-style hooks would be a bit cleaner.

In case you're curious, tmk/tmk_keyboard@53bd4a01be5c4327462e6c6620ea7670eddca6f2 was the commit that introduced hooks to TMK. There's more to that commit than just setting up hooks I believe, but the code for hooks looks pretty clean and fairly simple to move to QMK.

drashna, 2018-07-25:
This is partially done, now, actually. 
https://docs.qmk.fm/#/custom_quantum_functions?id=keyboard-idlingwake-code

- [ ]     Hook_late_init - last stage of init, just before keyboard loop (qmk = keyboard_init in #3113 
- [ ]     Hook_keyboard_loop - called from keyboard loop (often) (qmk = matrix_scan)
- [ ]     Hook_matrix_change - called matrix change event (keypress, often) (qmk = matrix_scan)
- [ ]     Hook_default_layer_change - default layer change event (qmk = default_layer_state_set, missing user function, has kb function)
- [ ]     Hook_layer_change - layer state change event (qmk = layer_state_set)
- [ ]     Hook_keyboard_leds_change - indicator LED change event (qmk = led_set)
- [ ]     Hook_bootmagic - called when checking bootmagic combo (qmk = TBD)

drashna, 2019-03-29:
I think the only thing that is left is bootmagic, actually.  And we have `bootmagic lite` which is user configurable, though not as fully featured.

jpetermans, 2019-04-04:
I haven't looked over everything in any detail, but there is no need for me to that. A quick review looks like it's all covered as you noted. Closing this issue. Thanks for the building in these additional options!

closed 2019-04-04 by jpetermans