qmk/qmk_firmware #1209: Blank name on device after flashing

rask, 2017-04-05:
I used QMK to flash by Satan PCB with a keymap. Now when plugged into a Linux Ubuntu computer and running `lsusb` the device name is blank. The ID is visible and fine, and the keyboard itself works fine.

When I use `lsusb -v` (verbose mode) the empty device reports `iManufacturer` as `SATAN` and `iProduct` as `GH60`, which are correct.

Is there a way to flash a device name properly for the keyboard? I would be fine just having something generic like the Satan GH60 product name visible in case a "custom name" is not possible.

RcColes, 2017-05-21:
This is probably a bigger issue than can be fixed unfortunately. The description from lsusb is populated from the file /var/lib/usbutils/usb.ids, which is in turn updated from the list from http://www.linux-usb.org/usb-ids.html.

In order to use that field we'd need to get the vendor ID and Product ID into the aforementioned list, which isn't possible because in order to do that we'd need a USB vendor ID, which are ~$5000. While it'd be really nice to have one this clearly isn't feasible.

One possible way around this is that you could manually add 0xFEED (QMK's unofficial USB vendor ID) and the product ID for the satan to the list, with the downsides that this would only make it display properly on your PC and that it'd likely get wiped with every major update to Linux.

fredizzimo, 2017-05-25:
I was going to write about this a while ago already, but forgot. I think we should apply a batch of PIDs from [pid.codes](http://pid.codes). As an open source project we should be able to do that. We should apply for at least the amount of keyboard that we have, but preferably for a few more, to prepare for new keyboards. 

After we have them, we can probably get them included in Linux, for Windows and Mac, we are most likely out of luck.

jackhumbert, 2017-05-25:
Yeah! @skullydazed and I were talking about that a little bit. I wanted to reach out to them to see if it made sense to grant QMK as a whole one PID, since a lot of the projects don't necessarily have open sourced hardware (something I think pid.codes requires).

From there, we could distribute the device versions among the keyboards we support, so that each one is unique. Some keyboards (like the Atreus) already have their own PID, so we could use those as well.

fredizzimo, 2017-05-26:
Getting just one PID, could be problematic, since the operating system could mess the devices up. That said, I don't think there has been any problems, even if it's the case with our made up VID/PID.

According to their site, the actual hardware doesn't necessarily need to be  open source, they say this [on the howto page](http://pid.codes/howto/)
> Containing schematics **or** source code for a device with a USB interface…

Furthermore 
> If your project involves both hardware and software, both need to be licensed under recognised OSS and OSHW licenses. If your project involves only one or the other, we may ask for further justification as to why you need a PID associated with your software project / development board instead of allowing end-users to request their own.

So the source code alone should be enough, I think QMK qualifies under *only one or the other* for most parts, with a few open source hardware devices as well, so we should be able to at least discuss with them about the options. In any case a separate discussion is probably needed anyway, since we would need a lot more PIDs than most other projects.

jackhumbert, 2017-05-26:
> Getting just one PID, could be problematic, since the operating system could mess the devices up

I'm kinda curious how computers would handle things with the same PID/VID, but different names/descriptions. I'm pretty sure changing the DEVICE_VER triggers the new hardware prompt (I could be mistaken on this).

> According to their site, the actual hardware doesn't necessarily need to be open source, they say this on the howto page

Yeah, I was looking at some of the accepted PRs, and they asked for PCB schematics _and_ layouts for some projects, and if we requested one for each project, that's not something we could provide (I don't publish the OLKB PCB layouts).

fredizzimo, 2017-05-26:
Windows might have problems https://www.osronline.com/showThread.CFM?link=173616. As can be seen [here](https://msdn.microsoft.com/en-us/library/windows/hardware/jj649944(v=vs.85).aspx), there are a lot of information that isn't classified by the bcdDevice version number.

It's probably fine in practice for keyboards, since all keyboards use the same default HID driver, but if custom drivers were used, I think we would have problems.

skullydazed, 2017-05-26:
If we progress to the point of providing custom drivers for {some,all} of our keyboards we can request a separate PID for that purpose. The tricky part there will be appropriately documenting that so people who come later will know not to use the default pid if they need a driver.

jackhumbert, 2017-05-26:
Ah, that's interesting - I wonder how that would handle the MIDI features. It'd be a pain to have to have a PID for each different driver configuration we'd need, but not impossible - I imagine automating this would be the way to go.

When @skullydazed and I talked about this earlier, he found [this](http://www.arachnidlabs.com/blog/2013/10/18/usb-if-no-vid-for-open-source/), which shutdown my dreams of acquiring a VID for QMK as a whole, but we may be in a different scenario, since we wouldn't be distributing them in the same way. It may be something worth talking to the USB-IF about, since they do have something for prototypes:

> We do have a vendor ID number designated for prototype products. This vendor ID number may not be used for a production product. Anybody who has such a need may contact us directly and we will provide them with the proto VID once they confirm that will not be making production products to be distributed and/or sold in the marketplace.

It'd also just be nice to get an official recommendation from them.

jackhumbert, 2020-03-13:
To loop back to initial question - if you modify the mentioned `usb.ids` with the name and VID/PID, this will show up on your system, but getting it on everyone else's system is a bit larger of an issue :) I'm gonna close this for now, but feel free to paste any code that you used for others' reference.

closed 2020-03-13 by jackhumbert