qmk/qmk_firmware #1210: Can't flash firmware to amj60

aizigao, 2017-04-05:
I'm use amj60, I had tried flash the qmk_firmware, but it didn't works

jpetermans, 2017-04-05:
Hi aizigao. You need to be more specific before anyone can help:
- what error messages are you seeing?
- firmware didn't compile correctly?
- compiles ok, but won't flash?
- flashes ok, but keyboard doesn't respond or keypresses are incorrect?
- what tools are you doing to flash
- using windows/linux/osx?
...

aizigao, 2017-04-06:
@jpetermans 
* I don‘t see any error massages
* complied correctly
* can flash , but the firm ware not works, keys is the  same before flashed. It seems not works
* I use tkg-tookit , simply change ghpad instead of amj60
* my system is macos
😞

aizigao, 2017-04-06:
@jeebak I just tried the qmk_firmware_flasher ,it works . I will close it ... 😄

closed 2017-04-06 by aizigao