qmk/qmk_firmware #1214: Need help on simple "chording" not "steno"

dunkarooftop, 2017-04-08:
I am wondering how to enable or write a function to make say

When press ```A and B``` keys together (with in 20ms) send ```CMD+C``` for copy

or when press ```A, B, and C``` keys together send ```Hyper+A```

Is this possible with QMK? Thanks :)

zelmon64, 2017-04-18:
@dunkarooftop This may have been added in #960 but I don't know how it's used. I just came across it (as well as this issue) because I was also wondering whether chording is possible with QMK (but I don't have a compatible keyboard yet to try it).

dunkarooftop, 2017-04-19:
During my research I only search "chording". Didn't think of searching "combo"  Thank you for pointing it out. I will give it a try and report back ! :)

dunkarooftop, 2017-05-02:
@zelmon64 
The combo function do work but with 1 limitation at the moment, you can not use the same character twice. More info on #960 

Here is my setup 4 combo to send various key combos.
```
// user defined combo
enum process_combo_event {
  WF_COPY, 
  RS_PASTE, 
  XC_PASTEBOT, 
  EI_LEADER, 
  };

const uint16_t PROGMEM copy_combo[] = {KC_W, KC_F, COMBO_END};
const uint16_t PROGMEM paste_combo[] = {KC_R, KC_S, COMBO_END};
const uint16_t PROGMEM pastebot_combo[] = {KC_X, KC_C, COMBO_END};
const uint16_t PROGMEM leader_combo[] = {KC_E, KC_I, COMBO_END};

combo_t key_combos[COMBO_COUNT] = {
    [WF_COPY] = COMBO_ACTION(copy_combo),
    [RS_PASTE] = COMBO_ACTION(paste_combo),
    [XC_PASTEBOT] = COMBO_ACTION(pastebot_combo),
    [EI_LEADER] = COMBO_ACTION(leader_combo),
};

void process_combo_event(uint8_t combo_index, bool pressed) {
    switch(combo_index) {
    case WF_COPY:
        if (pressed) {
                add_weak_mods(MOD_BIT(KC_LGUI));
                send_keyboard_report();
                register_code(KC_C);
                unregister_code(KC_C);
                del_weak_mods(MOD_BIT(KC_LGUI));
                send_keyboard_report();
        }
        break;

    case RS_PASTE:
        if (pressed) {
                add_weak_mods(MOD_BIT(KC_LGUI));
                send_keyboard_report();
                register_code(KC_V);
                unregister_code(KC_V);
                del_weak_mods(MOD_BIT(KC_LGUI));
                send_keyboard_report();
        }
        break;

    case XC_PASTEBOT:
        if (pressed) {
                add_weak_mods(MOD_BIT(KC_LALT));
                add_weak_mods(MOD_BIT(KC_LCTL));
                add_weak_mods(MOD_BIT(KC_LSHIFT));
                send_keyboard_report();
                register_code(KC_Z);
                unregister_code(KC_Z);
                del_weak_mods(MOD_BIT(KC_LALT));
                del_weak_mods(MOD_BIT(KC_LCTL));
                del_weak_mods(MOD_BIT(KC_LSHIFT));
                send_keyboard_report();
        }
        break;    

        case EI_LEADER:
        if (pressed) {
                add_weak_mods(MOD_BIT(KC_LALT));
                add_weak_mods(MOD_BIT(KC_LCTL));
                add_weak_mods(MOD_BIT(KC_LSHIFT));
                send_keyboard_report();
                register_code(KC_Y);
                unregister_code(KC_Y);
                del_weak_mods(MOD_BIT(KC_LALT));
                del_weak_mods(MOD_BIT(KC_LCTL));
                del_weak_mods(MOD_BIT(KC_LSHIFT));
                send_keyboard_report();
        }
        break;
    }
}
```

You also need to edit the value of ```#define COMBO_COUNT 4``` in ```config.h``` to match the amount of your combo. In my case is 4

I also edit the value of ```#define COMBO_TERM 20``` in ```quantum/process_keycode/process_combo.h```

The default value is ```TAP_TERM``` I made it ```20``` so only run the combo when it pressed at the same time pretty much.

drashna, 2018-03-25:
Combo/Chording needs actual documentation.

drashna, 2019-11-21:
Documentation for combos was added.   And I think the 2 character limit was resolved or has a PR open for it.

closed 2019-11-21 by drashna