qmk/qmk_firmware #1218: has_oneshot_mods_timed_out() is undefined when ONESHOT_TIMEOUT is set to 0

babarrett, 2017-04-09:
When building, for example, qmk_firmware/keyboards/ergodox/keymaps/townk_osx 
When I set ONESHOT_TIMEOUT to 3000 in config.h everything compile OK.
When I set ONESHOT_TIMEOUT to 0, indicating that I don't want my modifiers to cancel after a timeout, and the compile fails with:
```
Linking: .build/ergodox_ez_townk_osx.elf                                                            [ERRORS]
 |
 | .build/obj_ergodox_ez_townk_osx/keyboards/ergodox/keymaps/townk_osx/keymap.o: In function `matrix_scan_user':
 | /Users/bruce/dev/git/qmk_firmware/keyboards/ergodox/keymaps/townk_osx/keymap.c:270: undefined reference to `has_oneshot_layer_timed_out'
 | /Users/bruce/dev/git/qmk_firmware/keyboards/ergodox/keymaps/townk_osx/keymap.c:281: undefined reference to `has_oneshot_mods_timed_out'
 | collect2: error: ld returned 1 exit status
 |
make[1]: *** [.build/ergodox_ez_townk_osx.elf] Error 1
make: *** [ergodox-townk_osx] Error 1
```

Looks like an issue in qmk_firmware/tmk_core/common/action_util.c:
```
#if (defined(ONESHOT_TIMEOUT) && (ONESHOT_TIMEOUT > 0))
static int16_t oneshot_time = 0;
inline bool has_oneshot_mods_timed_out() {
  return TIMER_DIFF_16(timer_read(), oneshot_time) >= ONESHOT_TIMEOUT;
}
#endif
```
only defines has_oneshot_mods_timed_out() for values > 0, but has_oneshot_mods_timed_out() gets used in many places. 
We either need to define has_oneshot_mods_timed_out() when ONESHOT_TIMEOUT is zero or force the calls to has_oneshot_mods_timed_out() to not happen, throughout.

babarrett, 2017-04-10:
Same issue with: has_oneshot_layer_timed_out same file.

fredizzimo, 2017-04-10:
I made a fix for the `has_oneshot_mods_timed_out` as a part of this pull request #1217. But I didn't do anything about the `has_oneshot_layer_timed_out`, because it wasn't part of the visualization code.

drashna, 2018-03-25:
This appears to be fixed in #1405

closed 2018-03-25 by drashna