qmk/qmk_firmware #1219: Wiki missing important functions and keycodes

tusing, 2017-04-09:
I was looking through the **Fun with modifier keys** section [here](https://hub.docker.com/r/namaco/qmk_firmware/), and I noticed that the entire section is completely absent from the wiki.

Can we get it added to the wiki? Keycodes and examples like ```LALT(LCTL(KC_DEL))``` could save users a ton of time. Right now, if one were to just read the wiki as it is, people would think the only way to do this would be to write their own macros.

Furthermore, the "Fun with modifier keys" section has unique keycodes like ```KC_LPRN```, which are also entirely missing from the wiki.

jackhumbert, 2017-04-09:
Do you mean [this page](https://github.com/qmk/qmk_firmware/wiki/Key-Functions)?

tusing, 2017-04-09:
I see. My bad, I didn't think it would be under that section.

closed 2017-04-09 by tusing