qmk/qmk_firmware #1229: Stuck on 'Waiting for Bootloader...' Linux

Owimi, 2017-04-14:
Unable to reflash my keyboard. After holding down the reset button on my pcb nothing happens.

fredizzimo, 2017-04-14:
Hi @Owimi,

I think we need more information. For example. Which keyboard are you trying to flash? Does any LED turn on when you press the reset button?

Owimi, 2017-04-14:
I'm trying to flash a gh60 rev.c
No LEDs turn on when I press the button

Dbroqua, 2017-04-14:
Hello,

No led turn on when you press the flash button (in my memory).

To flash a gh60 you need (as root):
- connect the keyboard
- press flash button
- open terminal in `qmk_firmware/keyboards/gh60`
- sudo make keymap=default dfu

Example:
```
➜  gh60 git:(master) ✗ make keymap=dbroqua dfu     
WARNING:
 Some git sub-modules are out of date or modified, please consider runnning:
 git submodule sync --recursive
 git submodule update --init --recursive

 You can ignore this warning if you are not compiling any ChibiOS keyboards,
 or if you have modified the ChibiOS libraries yourself. 

Making gh60 with keymap dbroqua and target dfu

avr-gcc (GCC) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  20572	      0	  20572	   505c	gh60_dbroqua.hex

Compiling: keyboards/gh60/gh60.c                                                                    [OK]
Compiling: keyboards/gh60/keymaps/dbroqua/keymap.c                                                  [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/process_keycode/process_leader.c                                                 [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/light_ws2812.c                                                                   [OK]
Compiling: quantum/rgblight.c                                                                       [OK]
Compiling: ./tmk_core/common/host.c                                                                 [OK]
Compiling: ./tmk_core/common/keyboard.c                                                             [OK]
Compiling: ./tmk_core/common/action.c                                                               [OK]
Compiling: ./tmk_core/common/action_tapping.c                                                       [OK]
Compiling: ./tmk_core/common/action_macro.c                                                         [OK]
Compiling: ./tmk_core/common/action_layer.c                                                         [OK]
Compiling: ./tmk_core/common/action_util.c                                                          [OK]
Compiling: ./tmk_core/common/print.c                                                                [OK]
Compiling: ./tmk_core/common/debug.c                                                                [OK]
Compiling: ./tmk_core/common/util.c                                                                 [OK]
Compiling: ./tmk_core/common/eeconfig.c                                                             [OK]
Compiling: ./tmk_core/common/avr/suspend.c                                                          [OK]
Compiling: ./tmk_core/common/avr/timer.c                                                            [OK]
Compiling: ./tmk_core/common/avr/bootloader.c                                                       [OK]
Assembling: ./tmk_core/common/avr/xprintf.S                                                         [OK]
Compiling: ./tmk_core/common/bootmagic.c                                                            [OK]
Compiling: ./tmk_core/common/mousekey.c                                                             [OK]
Compiling: ./tmk_core/common/avr/sleep_led.c                                                        [OK]
Compiling: ./tmk_core/protocol/lufa/lufa.c                                                          [OK]
Compiling: ./tmk_core/protocol/lufa/descriptor.c                                                    [OK]
Compiling: ./tmk_core/protocol/lufa/outputselect.c                                                  [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Class/Common/HIDParser.c              [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c               [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c       [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c             [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c                 [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c           [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c                 [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c        [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c         [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/ConfigDescriptors.c              [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/DeviceStandardReq.c              [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/Events.c                         [OK]
Compiling: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/HostStandardReq.c                [OK]
[... cut ...] ng: ./tmk_core/protocol/lufa/LUFA-git/LUFA/Drivers/USB/Core/USBTask.c                        [OK]
Linking: .build/gh60_dbroqua.elf                                                                    [OK]
Creating load file for Flash: .build/gh60_dbroqua.hex                                               [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  20572	      0	  20572	   505c	gh60_dbroqua.hex

Bootloader Version: 0x00 (0)
Validating...
20572 bytes used (71.75%)
```

Owimi, 2017-04-14:
'Error:  Bootloader not found. Trying again in 5s'

fredizzimo, 2017-04-14:
The readme says that it should be flashed with the teensy loader. Have you checked the following instructions? https://www.pjrc.com/teensy/loader_linux.html

Once that's setup it should work the same way as the `dfu` command, but replace `dfu` with `teensy`

Dbroqua, 2017-04-14:
Usually if you read this error it's because you haven't pressed the reset button.

Owimi, 2017-04-14:
Oh, It also says 'dfu-programmer:no device present.'

Dbroqua, 2017-04-14:
Install missing programs and it will be ok :D .

On my debian: 
```
dfu-programmer					install
dfu-util					install
```

Owimi, 2017-04-14:
I think those programs were already installed but I tried to install them again and nothing seems to have changed

Owimi, 2017-04-15:
Is it possible that I am missing needed drivers?

fredizzimo, 2017-04-15:
Did you try the teensy loader and the instructions I gave above?

The [GH60 readme](https://github.com/qmk/qmk_firmware/tree/master/keyboards/gh60) tells you to use that and not the `dfu-util` or `dfu-programmer`.

Owimi, 2017-04-15:
When I run teensy, the popup window tells me to press the button on my pcb, but when I do nothing happens. To the bottom of the same window there's a text saying 'No file specified'

Dbroqua, 2017-04-15:
Are you sure: 
* keyboard correctly plugged
* teensy run as root (or with sudo)

Which GNU/Linux are you using? Installed in Virtual environment or physical system?

Owimi, 2017-04-16:
Running Ubuntu with Virtualbox on Windows 10.
Running teensy as root does not seem to help. 
In the teensy box it says 'No file specified'.

Owimi, 2017-04-16:
I think I have figured out my problem. I am using the wrong kind of usb cable. I need a proper mini usb cable that actually transfers data, not just power.

Dbroqua, 2017-04-16:
Hello,

In virtualbox have you selected the keyboard in the list of available USB devices ?

http://2.bp.blogspot.com/-_KM7KnTPKMM/UZkc0kPocSI/AAAAAAAADMU/2XwBF1T9MAU/s1600/mount+usb+drive+on+virtualbox.png

If you don't select the GH60 is the above list your Ubuntu system can't communicate with it.

Owimi, 2017-04-17:
Finally managed to flash the board. I think the problem might have had to do with virtualbox using usb 1.0 instead of 2.0.
Thanks you guys so much for all the help.

closed 2017-06-24 by skullydazed