qmk/qmk_firmware #1230: Infinity Ergodox using simple_visualizer.h, slave LCD stops showing text, but background color still responsive.

r2d2rogers, 2017-04-14:
After more than an hour plugged in and active, with Win10 machine not suspended or rebooted, only locked and unlocked, my slave(right side in this case) LCD stops displaying the text showing the layers and NUM, CAPS, etc. text.  The background lighting stays responsive to layer and CAPSLOCK state changes.

My code and keymap are on the r2d2rogers branch of my fork of qmk_firmware.

northrup, 2017-04-16:
My follower keyboard side on MacOS goes completely inactive after hour. No text, no colors, no nothing. This behavior is happening after the recent merges for LCD support for the Ergodox Infinity.

fredizzimo, 2017-04-17:
This will be very hard to track down, since I have not seen it happen on my keyboard. And debugging it would require a hardware debugger anyway.

I guess since you are using the simple_visualizer, there are no animations or anything happening until it goes dark? Does it happen without touching the keyboard, or is it something that happens as a response to something that you do? 

It would also be nice to see what happens just a fraction before it stops working. It's probably hard to catch that moment with bare eyes, but maybe it would be possible to setup a video recording.

My current theory is that its a power issue, causing the LCD to shutdown, but keeping the CPU running. On the other hand the report by @northrup, seems to indicate that's not the case for him. It would also be nice to know if the problem goes away if the slave is disconnected and re-connected. But I'm not sure if the slave is electrically hot-pluggable or not, so it could be risky to try that.

r2d2rogers, 2017-04-17:
Just back from lunch, and the right hand unit was as I described above.  I unplugged from the left side ans reconnected, and it came up with the logo, and is back to displaying the layer info with the simple_visualizer.  Both sides were responsive as I typed before I unplugged the right, and the layer switching from the right functioned correctly.

fredizzimo, 2017-04-17:
Thanks @r2d2rogers,

At least it means that the state of the master is correct, so it doesn't send something bad to the slave. 

Could you try to disable [selective suspend](http://www.techsupportalert.com/content/how-fix-annoying-windows-usb-problem.htm) and see if that helps?

r2d2rogers, 2017-04-17:
@fredizzimo, selective suspend is currently disabled on this machine.
[image]

fredizzimo, 2017-04-17:
I don't think the power plan should change by itself, unless there's some manufacturer software that does something strange.

So this rules out yet another thing to look for.

One thing that probably would help in debugging would be to contiguously measure the voltage to the slave. But that would require a digital oscilloscope, or for example a Teensy, Arduino, or Raspberry PI running the right software sampling through the ADC input. To measure the +5V line a voltage divider would probably be needed. The next step would be to use a logic analyze to check the actual communication between the halves, that's also something the above boards could easily do, if the communication speed is dropped low enough.

But I'm probably hoping too much finding someone with the problem, having the hardware and being able to do that.

r2d2rogers, 2017-04-17:
I see that the issue had occurred again in the last 90 minutes, so that doubly rules out the selective suspend. 
I do have two Raspberry Pi's sitting at home, were you thinking to have the keyboard plugged into the Pi and have the pi monitoring it's own voltage?  I can see if I get the same behavior on one of those tonight.  I may need a suggestion or link on how to set up the voltage monitoring...

fredizzimo, 2017-04-17:
Since I'm a software guy it's better if someone else could explain the actual connection needed. But in short you want to measure the voltage between vss and +5 and those are exposed as test points at the bottom of the PCB.

The bad news is that I did some searches on the net, and it seems like I was wrong about Raspberry PI and it doesn't have a built in ADC. It's possible to do it with that, but it either requires an add-on module or chip.

lisinge, 2017-04-17:
This happens to me too. But only on the left side, no matter if it's the master or the slave.
The keyboard is still responsive and the backlight of the LCD still works and animating properly, it just wont show any text.

I'm running windows and i've tried to disable the USB suspending setting.

It has recovered by itself once but then the whole of the LCD is covered with semi active pixels like this:
[image]

r2d2rogers, 2017-04-18:
@fredizzimo, This morning, I have the two halves plugged in independently, and the right side has just gone blank, even though it should be acting as master.

lisinge, 2017-04-18:
In my earlier comment when i reported about this with the picture i was using Windows.

Today i've had my keyboard plugged in to my Mac the whole day while working and i haven't experienced the issue once.

Could this be OS related? I know @northrup mentioned he was using mac but his slave half seems to be dying completely, whilst all of ours just seem to stop displaying the text on the LCD.

@r2d2rogers are you using only Windows or are you seeing this on Mac too?

r2d2rogers, 2017-04-18:
I'm only seeing this specific behavior on Windows, the computer at home is Linux, and sat overnight without the text disappearing.

fredizzimo, 2017-04-18:
Hm.. The operating system theory is interesting. But needs more data points to draw any conclusions. It might also at least be partly hardware dependent, at least from judging @r2d2rogers experiment last night.

For the information, I'm using windows and I'm not experiencing the issue.

Have any of you been trying the official Input Club firmware? Does that behave the same way? 

Another thing that feels strange is that the issue happens with the simple_visualizer, which does nothing unless you activate new layers or caps lock state for example. In this situation the firmware does not send any commands to the LCD screen, so it doesn't feel like a software bug in the communication code.

lisinge, 2017-04-18:
I used the Input Club firmware for a couple of weeks just up until the other day when i started messing aroudn with QMK when you fixed the LCD support. I didn't experience any problems at all with inputclub's firmware.

Another thing that i find interesting is that i experience this on my left half regardless if it's master or not while @r2d2rogers has this behaviour on his right half

fredizzimo, 2017-04-18:
The fact that it happens on the master as well as slaves make the debugging a little bit easier.

Would it be possible for you to add some debug output to `write_data` in `/keyboards/ergodox/infinity/drivers/gdisp/st7565ergodox/board_ST7565.h`. Then you would at least be able to see if the keyboard tries to communicate with the LCD just before it goes blank.

If it's setup correctly, you should be able to read the messages with HIDListen.  I think the documentation for `CONSOLE_ENABLE` in https://github.com/qmk/qmk_firmware/wiki/Make-Instructions, are the best instructions we have at the moment.

r2d2rogers, 2017-04-18:
I have a meeting in a few minutes, but just noticed the two rows of text are now upside down!  Plugged in right side alone, still responsive, but each in it's own row, vertically flipped. picture to follow

r2d2rogers, 2017-04-18:
I have a meeting in a few minutes, but just noticed the two rows of text are now upside down! 
[image]

lisinge, 2017-04-18:
> but just noticed the two rows of text are now upside down

The exact same thing hapened to my left side, but it has only hapened once so far

r2d2rogers, 2017-04-19:
Bad news for the OS theory, I just had the same symptom on the Linux machine at home.

fredizzimo, 2017-04-19:
I have some good news. I found a problem in the LCD pin configuration, and fixed it in #1245. 

I don't know if it's placebo effect or what it is, but even with the wrong code before my new fix I have personally not seen any issues. So my previous change to the code seemed to fix things for me, but obviously not for you.

I don't think it could have though as I'm pretty sure the typo made the the pin get an "unconnected" configuration and therefore floating or random state. 

Hope that this fixes the problem.

lisinge, 2017-04-19:
I will give it a try right away!

lisinge, 2017-04-19:
I've been running with the fix from #1245 for a couple of hours. And i haven't seen the error once. 
I can't say for sure but it looks like it fixed it because before it happened within an hour of plugging in the keyboard.

Thanks @fredizzimo!

fredizzimo, 2017-04-19:
Great, I got the clue from @r2d2rogers, upside down screen. 

The only way that could happen is by sending the "Common output mode select" command to the LCD screen. It's only sent from one place in the code, during the startup. And in that case the value is hardcoded to the right direction.

So I figured out it there has to be some electrical interference causing the command to be sent. Because the Input Club firmware works, I also figured out that it has to be a software issue. 

So I started to go through my code yet another time, and saw the typo. 

The pin in question is the chip select pin, which enables the communication with the chip. Electrical noise would cause it to be active when it shouldn't. Then other noise on the actual communication pins would make it send random commands all the time. Some of those shuts the LCD down completely, while others could cause the LCD to flip for example.

r2d2rogers, 2017-04-19:
Sweet! I have the latest code pulled, compiled, and flashed to both sets, I'll watch it a while to confirm and close this out.

lisinge, 2017-04-19:
@fredizzimo Thanks for the explanation, i'm new to this so it's super interesting to know these things

r2d2rogers, 2017-04-19:
Looks great, thanks @fredizzimo!

closed 2017-04-19 by r2d2rogers