qmk/qmk_firmware #1232: Unable to initialize libusb: -99

YamiFrankc, 2017-04-16:
Hello,

I am using the Ubuntu environment inside of Windows 10 to flash my Infinity Ergodox, and just when i do `sudo make infinity-keyboard-dfu-util` , i get an error code that says "Unable to initialize libusb: -99" and I cant figure out what is happening, any ideas?

fredizzimo, 2017-04-16:
The Windows subsystem for Linux unfortunately doesn't support USB properly so you can't flash the device from there. You can either run dfu-util manually from a normal windows command prompt, or setup the build environment in mingw or cygwin. Follow the Vista or later instructions here https://github.com/qmk/qmk_firmware/wiki/Build-Environment-Setup. 

With the new windows 10 creators update it should also be possible to run normal windows programs directly from the WSL command prompt. I haven't tested it myself, but it should be possible to set it up to run the normal windows dfu-util instead of the Ubuntu one. If this works, then it would be possible to do everything from within the same environment. 

If you feel like experimenting, you could try to do it that way. In any case I will try it soon, probably during next week, and update the instructions if it turns out successfully.

YamiFrankc, 2017-04-16:
Thanks for your fast reply.
I tried with `dfu-util -D file.hex` while the keyboard was on flash mode,  and it loads and finishes the command without errors("Unable to read fu status after completion" is the last line of output), but the keyboard stays in flash mode even after I unplug it.

That was after following the Vista instructions,  but im not sure if it uses a special method to flash? It doesnt say anything on how to do it after building the environment.

Lastly, I used the GUI flash tool from Input Club, but using the .hex file. Again, it seems to work fine but the keyboard just ends up with no firmware

fredizzimo, 2017-04-16:
"Unable to read DFU status after completion" is normal, at least that's what it says for me.

After flashing, the keyboard should automatically reboot and start working, without any need to plug it out.

Does it work with a hex file from Input Club? Maybe you have some problem in the keymap that causes it to crash and enter bootloader mode.

YamiFrankc, 2017-04-17:
They use .bin files to flash the keyboard, but yest it works correctly.
I also tried using the official QMK firmware flasher, but it wont allow me to click on Flash.

YamiFrankc, 2017-04-17:
I want to give a update:
Following this [https://github.com/qmk/qmk_firmware/issues/575](url) i was able to get a .bin file that I can flash usin the normal infinity method.
So it works now!

closed 2017-05-24 by fredizzimo