qmk/qmk_firmware #1233: KC_LPRN + Right Shift makes the keyboard go crazy

jdavis, 2017-04-16:
Hello!

I noticed a really strange behavior where if I push my [right shift](https://github.com/jdavis/ergodox/blob/master/keymap.c#L69) and combine it [with `(`](https://github.com/jdavis/ergodox/blob/master/keymap.c#L59), the following happens:

* https://twitter.com/HopefulJosh/status/853478674032541696

Keys I push dump multiple numbers instead of the desired key.

Example:
```
KC_J: 7f29
KC_K: 4000
KC_O: 3FDE
```

At first I thought it was due to [my macro](https://github.com/jdavis/ergodox/commit/37f42fa0b62b30c0cda57433958a1c7bb4b27367#diff-c507ac09c450ed169ea35b9b88daa9c7L259), but even after disabling it, it still happens.

Any ideas on debugging this?

jdavis, 2017-04-16:
Additionally:
* I haven't found another key that triggers this other than `KC_LPRN`.
* I just switched `KC_RSFT` to `KC_LSFT` and it doesn't happen anymore.

zhou13, 2017-06-24:
I can confirm this bug.  I found this when I tried to enable space cadet rollover: after pressing two shift simultaneously, my keyboard went crazy.

jayliu50, 2017-06-28:
I recall this happening to me as well. 

I think you are inadvertently activating the MAGIC modifier. Notice that `(` is (probably) shorthand for LSFT(KC_9). Magic keys by default are LSFT + RSFT. 

You might consider this change in your config.h, and see if that presents that from happening:

```c
/* key combination for magic key command */
#define IS_COMMAND() ( \
    keyboard_report->mods == (MOD_BIT(KC_LSHIFT) | MOD_BIT(KC_RSHIFT)) \ // <-- right here, change this to something other than LSHIFT | RSHIFT
)
```

jackhumbert, 2017-06-29:
I think @jayliu50 is right-on here - @ezuk you may want to consider something like their suggestion! I'm gonna close this, but if this doesn't work for someone, please reopen/comment!

closed 2017-06-29 by jackhumbert

chyczewski-maciej, 2018-07-14:
I got the same issue. Is there any other solution to that?