qmk/qmk_firmware #1237: A couple questions/issues

YamiFrankc, 2017-04-18:
Hello,

I am using an Infinity Ergodox.

I have been able to add new and functional layers, but when I activate them the screens on the keyboards do not change, and I cant find the part where to set the color and text of layers. Where is it? Im guessing I would also be able to modify the time the screen takes to change colors, right?

Sometimes the screen on the right half(I have the left half plugged into the computer, and the right in the left one) will stop displaying text but it still changes colors. Not a great deal for me but maybe its a bug.

I was reading the qmk wiki and for some features like tap_dance you need to modify the makefile, but there are many (one inside the ergodox folder, one inside ergodox/infinity, and one in the root folder). Where should I make these configs?

And last, I feel like many of my questions are not supposed to be asked here. Is there a qmk forum or something?

Thanks

r2d2rogers, 2017-04-18:
See the active discussion on #1230 about the screen on the right going blank.

As far as the layer labels are concerned, your keymap directory should have a visualizer.c file for controlling the LCDs.  This support was just added by @fredizzimo and is still in need of better documentation.  Try reading the files in the default keymap directory, and the keyboards/ergodox/infinity/simple_visualizer.h file to see where to add your own layer labels.

YamiFrankc, 2017-04-18:
Thanks, I added 
`    else if (state->status.layer & 0x6) {
        state->target_lcd_color = LCD_COLOR(268, saturation, 0xFF);
        state->layer_text = "Dvorak";
    }`
to visualizer.c and it compiles but the screen wont change colors still. I took other entry as an example and edited the 0x4 for a 0x6, which seemed to be the logical progression.

r2d2rogers, 2017-04-18:
Actually it should be 8.  Default, 2,4,8,10,20,40,80,100,200 etc.

I'm using up to 20, layer 5 (default is 0).

Do you have 4 layers defined?

YamiFrankc, 2017-04-19:
Thanks, so it goes doubling numbers, right?
Yes, I have the three default ones and an added dvorak one

fredizzimo, 2017-04-19:
Yes, sorry for using hex numbers in the example code, I really shouldn't have done that and will fix it at some point.

Something like 
```
if (state->status.layer & (1u << 3)
else if (state->status.layer & (1u << 2)
else if (state->status.layer & (1u << 1)
else if (state->status.layer & (1u << 0)
```
would probably have been better. But even better would be a macro as suggested at the bottom of #1217.

The reason why it's you need power of two's (doubling the numbers), is because you are doing bit testing. Layers in QMK are represented by bits i in the `layer` and `default_layer` variables being turned on, and there can actually be many layers active at the same time, so we can not use normal comparison.  It's clearer if we convert the numbers to binary, so we have 
0001 = 1 = first layer active
0010 = 2 = second layer active
0100 = 4 = third layer active
1000 = 8 = fourth layer active

To test the if a layer is active we  do an `and` which returns true (a non-zero value) if a bit in the same place are both one, for example, let's say layer 3 is active
```
0100(4) & 0100(4) = 0100 = true
0100(4) & 1000(8) = 0000 = false
```
The reason for doing the test from the highest to lowest is because when multiple layers are active, you typically want to know the highest one. So if layer 2 and 3 are active
```
0110 & 0100 = 0100 = true
0110 & 0010 = 0010 = true
```

fredizzimo, 2017-05-24:
Closing this as the issue appears to be solved.

closed 2017-05-24 by fredizzimo